feilipu/Arduino_FreeRTOS_Library #3: appears that String variables are not being printed correctly

rkertesz, 2016-06-27:
I can't seem to get a string variable printed to monitor when using FreeRTOS. works fine in regular wiring language.
e.g. from analogread_digitalread example:
only blanks print to screen
```

  for (;;)
  {
    // read the input on analog pin 0:
    String sensorValue = String(analogRead(A0));

    // See if we can obtain or "Take" the Serial Semaphore.
    // If the semaphore is not available, wait 5 ticks of the Scheduler to see if it becomes free.
    if ( xSemaphoreTake( xSerialSemaphore, ( TickType_t ) 5 ) == pdTRUE )
    {
      // We were able to obtain or "Take" the semaphore and can now access the shared resource.
      // We want to have the Serial Port for us alone, as it takes some time to print,
      // so we don't want it getting stolen during the middle of a conversion.
      // print out the value you read:
      Serial.println(sensorValue);

      xSemaphoreGive( xSerialSemaphore ); // Now free or "Give" the Serial Port for others.
    }

    vTaskDelay(1);  // one tick delay (15ms) in between reads for stability
  }
```

feilipu, 2016-06-28:
rkertesz,

Just retested a few different sketches (including both example sketches) with a clean install of Arduino 1.6.9 and the 8.2.3-14 Arduino_FreeRTOS library version, on Windows 10,

It works as expected. I can't replicate your issue.
Perhaps provide some more background information.

rkertesz, 2016-06-28:
Really? That is surprising. Did you cast the result of the analog read int to a string as I showed in the example code above?
I am using the latest arduino IDE as well. I am actually heartened to see that it works but I tried so many different ways to use String.

Thank you

feilipu, 2016-06-28:
I've tested the one line change, and confirmed that there is an issue with the String library.

```
    int sensorValue = analogRead(A0); // Original, works correctly
    // String sensorValue = String(analogRead(A0)); // Proposed, prints blanks
    // String sensorValue = "hello"; // Tested, also prints blanks
```

As `hello` doesn't print either, the problem doesn't originate from the analogRead library.

I don't have time to dig into `String()` now to find out why it is broken. But I'll have a look soon.

rkertesz, 2016-06-28:
Ah. Thank you. Yes. I agree that it is a pervasive String issue. Thank you for double checking this. I just wanted to note that it looks like it isn't just a printing issue but the string variable doesn't get populated with any text. A *char array does work as expected.

feilipu, 2016-06-28:
I'll take a look at String() shortly, to see if there is anything that can be fixed in that library.

closed 2016-06-28 by feilipu

rkertesz, 2016-07-06:
Should I just keep an eye on the master branch for this fix?

feilipu, 2016-07-07:
Yes. Best place to look, together with a 9.0.1 version, once it is released.

But note that there is little chance that I'll be able to influence the Arduino developers to change the String library, unless a general issue can be proven.

Otherwise, just use char* as the workaround.

feilipu, 2016-07-22:
I've spent some time identifying the limitations here, and have the following input.
String functions seem to be very finicky. But the following code segments deliver what you need.

```
#include <Arduino_FreeRTOS.h>

void TaskAnalogRead( void *pvParameters );

String sensorValue; // Make the sensorValue a Global Variable, otherwise it will be reallocated each time the Task for() loop is run.

// the setup function runs once when you press reset or power the board
void setup() {

  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);

  sensorValue.reserve(255); // Reserve a fixed space for your String, larger than the maximum length required, to avoid reallocation.

  // Now set up a Task to run independently.
  xTaskCreate(
    TaskAnalogRead
    ,  (const portCHAR *) "AnalogRead"
    ,  128  // Stack size
    ,  NULL
    ,  1  // Priority
    ,  NULL );

  // Now the Task scheduler, which takes over control of scheduling individual Tasks, is automatically started.
}

void loop()
{
  // Empty. Things are done in Tasks.
}

/*--------------------------------------------------*/
/*---------------------- Tasks ---------------------*/
/*--------------------------------------------------*/

void TaskAnalogRead( void *pvParameters __attribute__((unused)) )  // This is a Task.
{
  for (;;)
  {
    // read the input on analog pin 0:

    int analogValue = analogRead(A0); // Embedding functions in String() doesn't work well, so separate the function call out.
    sensorValue = "is: " + String( analogValue ); // Concatenation is needed, or the String conversion seems to fail.

    // sensorValue = String( analogValue );
    // sensorValue = String(analogRead(A0));
    // String sensorValue = String(analogRead(A0), DEC);

    Serial.print("Address: ");
    Serial.print(int(&sensorValue), HEX);
    Serial.print("  Length: ");
    Serial.println(sensorValue.length());

    // print out the value you read:
    Serial.print("Sensor Value ");
    Serial.println(sensorValue);

    vTaskDelay(10);  // one tick delay (150ms) in between reads for stability
  }
}
```