feilipu/Arduino_FreeRTOS_Library #4: Unable to read from SD inside of tasks

sjkelleyjr, 2016-07-17:
When I create a task to read from the SD card using:

File myFile = SD.open("temp.txt",FILE_READ);

The file can't be opened.  I tried placing noInterrupts(); in various places with no luck, as well as increasing the stack size for the task.

I can both read and write from the same file in the setup, the issue only occurs when placing the code in a task.

Sailanarmo, 2016-07-19:
I can confirm, 

If I run the following piece of code I can write to the SD Card. If I run it in a task, it will create the text file, however it will not write anything to it. 

`File myBaro = SD.open("test.txt", FILE_WRITE);
 baro.readBaro();
 Serial.println(baro.getPressure());
 myBaro.println("This should be working.");
 myBaro.println(baro.getPressure());
 delay(200);
 myBaro.close();`

Again, in setup it works 100%, but in a task for whatever reason it will not write to the SD Card.

feilipu, 2016-07-19:
Been unwell last few days.
Will update when back on code.
Sorry.

sjkelleyjr, 2016-07-20:
It looks like we got it working by opening the file in the setup and leaving it open in the task.  IE, the file should already be opened by the time the task gets a hold of it.

feilipu, 2016-07-21:
That's good to hear.

I'm certain there is a String() class issue underlying this issue though.

The only avenue I have to investigate is the potential for a variable declaration / freeing issue in the difference between the Arduino implementation, where setup() and loop() are both called by the main() function and hence may have corresponding variable space, and the FreeRTOS case where main() is never called directly. But, I haven't nailed down anything concrete yet.

closed 2016-07-21 by feilipu