feilipu/Arduino_FreeRTOS_Library #7: serialEvent not handled

asharect, 2016-08-30:
I used arduino example "SerialEvent", it was ok, but when I include <Arduino_FreeRTOS>, the serialEvent didn't work.  how to fix it? use avr ISR?

feilipu, 2016-08-30:
I have tried to replicate the issue you describe, but I can't.
See below screenshot.
[image]

The AVR ATmega is a very simple MCU. There is no interrupt prioritisation system (except intrinsic ordering). The configMAX_SYSCALL_INTERRUPT_PRIORITY option is not relevant to an ATmega port.

Perhaps you can reconfirm the issue?

closed 2016-08-31 by feilipu

joe31093, 2016-12-17:
I see this is closed but I have the code running on a Teensy 3.6.  I have two tasks running.  Both are at idle + 1 priority.  One task toggles the LED on Pin 13 and the other is waiting for a semaphore give when the SerialEvent() is called.  I have a print in the serial event and it is not printing anything out.  I tested it the same way in the code you have above and it works fine.  Looks like running the scheduler affects SerialEvent.

joe31093, 2016-12-17:
I am not sure if this helps.  I recently installed 8.2.3 (greiman) on my Teensy 3.6.  The serial event is not being call.  I have two tasks running.  A heartbeat task which turns on off the LEDs.  Another task waiting on a sem give in the serial event function so in can process the text.  I have a print statement in the serial event and it is not being printed.  I ran a similar example of the code you have above and the serial event worked fine.  I am guessing running the scheduler affected the system interrupts.