feilipu/Arduino_FreeRTOS_Library #9: Pro Micro and the Hackster.io Analog Read w/Blink...

SalvorinFex, 2016-09-19:
I have been trying various times to get the Pro Micro to use the code for FreeRTOS on this page:
(https://www.hackster.io/feilipu/using-freertos-multi-tasking-in-arduino-ebc3cc)

After some instruction from Phillip, I have modified some of the code:

```
#include <Arduino_FreeRTOS.h>

// define two tasks for Blink & AnalogRead
void TaskBlink( void *pvParameters );
void TaskAnalogRead( void *pvParameters );

// the setup function runs once when you press reset or power the board
void setup() {

  // Now set up two tasks to run independently.
  xTaskCreate(
    TaskBlink
    ,  (const portCHAR *)"Blink"   // A name just for humans
    ,  128  // Stack size
    ,  NULL
    ,  2  // priority
    ,  NULL );

  xTaskCreate(
    TaskAnalogRead
    ,  (const portCHAR *) "AnalogRead"
    ,  128 // This stack size can be checked & adjusted by reading Highwater
    ,  NULL
    ,  1  // priority
    ,  NULL );

  // Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
}

void loop()
{
  // Empty. Things are done in Tasks.
}

/*--------------------------------------------------*/
/*---------------------- Tasks ---------------------*/
/*--------------------------------------------------*/

void TaskBlink(void *pvParameters)  // This is a task.
{
  (void) pvParameters;

  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);

  for (;;) // A Task shall never return or exit.
  {
    digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
    digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
  }
}

void TaskAnalogRead(void *pvParameters)  // This is a task.
{
  (void) pvParameters;

  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
   // while the serial stream is not open, do nothing:
   while (!Serial) ;

  for (;;)
  {
    // read the input on analog pin 0:
    int sensorValue = analogRead(A0);
    // print out the value you read:
    Serial.println(sensorValue);
    vTaskDelay(1);  // one tick delay (15ms) in between reads for stability
  }
}
```

Namely adding the `while  (!Serial);` towards the end of the sketch.
Which now allows it to compile and upload, but the board does not send anything over serial and does not blink the LED.

This is where I am right now and request further help.

Thank you!

feilipu, 2016-09-20:
What happens if you leave out the `while(!Serial); `statement?

As background, the Leonardo and Micro (32u4) family with the integrated USB interface impose a significant background processing load on the AVR platform, related to keeping the USB interface alive. There is exact ms timing required on responses on USB, so (any) Libraries that disable the global Interrupt on these devices can have issues.

The Arduino IDE disguises this background USB processing from the user.

I suggest the the following is happening. Since the 32u4 is running a background timer interrupt (with further interrupts disabled during servicing) to respond to 1ms USB messages on the Serial port, and you've called Serial within a Task (which disables interrupts during the Scheduler activity) the two items are deadlocking.

By deadlock I mean that the 32u4 needs to respond to USB messages to keep the interface alive, but the Scheduler is running too and therefore the USB interrupt cannot be serviced in a timely manner, and therefore the Serial interface is never established, ensuring that the `while(!Serial);` statement is never completed. Deadlock!

As a work around, try setting up the Serial communications in the setup() function. This is before the FreeRTOS Scheduler is enabled, so the conflict will never occur.

Also, don't hammer the Serial() command so hard, by sticking it in a while() without delay.
Try;
```
while(!Serial)
    TaskDelay(1);
```

SalvorinFex, 2016-09-20:
I didn't have the `while` in there until you suggested it, when it also wouldn't work correctly.
So, this is what it looks like now:
```
void setup() {

  // Now set up two tasks to run independently.
  xTaskCreate(
    TaskBlink
    ,  (const portCHAR *)"Blink"   // A name just for humans
    ,  128  // Stack size
    ,  NULL
    ,  2  // priority
    ,  NULL );

  xTaskCreate(
    TaskAnalogRead
    ,  (const portCHAR *) "AnalogRead"
    ,  128 // This stack size can be checked & adjusted by reading Highwater
    ,  NULL
    ,  1  // priority
    ,  NULL );

while(!Serial)
    TaskDelay(1);
  
  // Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
}
```
But I am confused about the `Task` portion of this command, since it's in the setup, I am guessing the Tasks have not yet been assigned to the scheduler, yet.

Which is a believe why this error, now occurs:
```
C:\Users\kroesener\Documents\Arduino\FreeRTOS_testing2\FreeRTOS_testing2.ino: In function 'void setup()':

FreeRTOS_testing2:28: error: 'TaskDelay' was not declared in this scope

     TaskDelay(1);

                ^

exit status 1
'TaskDelay' was not declared in this scope
```

feilipu, 2016-09-20:
Yes. Before the Scheduler is started (with the hidden code between `startup()` and `loop()`) the TaskDelay function will not work (because the Scheduler isn't running and therefore can't schedule a delay). So in the `startup()` function the traditional form of `while(!Serial());` can be used.

SalvorinFex, 2016-09-20:
Alright, some progress.
I had to interpret `startup()` to `setup()` and I added this:
```
  // Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
while (!Serial) {
  ;
}
}
void loop()
{
  // Empty. Things are done in Tasks.
```
Now I do get serial `AnanlogRead` numbers in the Serial Monitor.
No blinking pin 13 though.

I am kinda of curious, since these 32u4 boards are supported by FreeRTOS as well as this project, were there tests run to make sure this code ran on these boards?

feilipu, 2016-09-20:
You may need to refer to the LED as `LED_BUILTIN` although in the [`pins_arduino.h`](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/leonardo/pins_arduino.h) file it does default to 13. If you're getting output in the Serial Monitor, then from the FreeRTOS point of view it is all good. Try printing debugging info to see where your code is working, and where not.

The 32u4 is a normal ATmega MCU, and the FreeRTOS port is running successfully.
So, its supported.

The issue, which applies to everything in the Arduino world, is working around what the IDE does for you and to you to try to make things "easy". The FreeRTOS Library works around lots of Arduino IDE created capabilities (restrictions), but can't work around the fact that the ATmega MCU only has one interrupt priority available.

SalvorinFex, 2016-09-20:
Heh, I was also, just about to post, everything works fine on my Leonardo and Micro, then I realized the pins for the LEDs on the Pro Micro are designated RXLED and TXLED.

So, with that last modification, they are working normally now.

Thank you for your help, it is much appreciated!

closed 2016-09-20 by feilipu