feilipu/Arduino_FreeRTOS_Library #11: Blink Not Working On Raspbian Arduino IDE Version

dhamukrish, 2016-12-18:
Could you please help me what is wrong in the below code with Raspbian version of Arduino IDE as I could not see build in LED flickering but continuous. But the same code works fine from MacOS version of Arduino IDE.

Please advise.

```
#include <Arduino_FreeRTOS.h>

void TaskBlink( void *pvParameters );

void setup() {

  xTaskCreate(
    TaskBlink
    ,  (const portCHAR *)"Blink"
    ,  128
    ,  NULL
    ,  2
    ,  NULL );
}

void loop() {
}

void TaskBlink(void *pvParameters) {
  (void) pvParameters;

  pinMode(LED_BUILTIN, OUTPUT);

  for (;;)
  {
    digitalWrite(LED_BUILTIN, HIGH);
    vTaskDelay( 1000 / portTICK_PERIOD_MS );
    digitalWrite(LED_BUILTIN, LOW);
    vTaskDelay( 1000 / portTICK_PERIOD_MS );
  }
}
```

NicoHood, 2016-12-18:
You need to download the latest arduino ide from arduino.cc for arm. The raspbian software is outdated.

dhamukrish, 2016-12-19:
@NicoHood I can see now working after installed the latest version. Thanks for your support.

feilipu, 2016-12-19:
Good to hear!

closed 2016-12-19 by feilipu