feilipu/Arduino_FreeRTOS_Library #12: Invalid Library message...

SalvorinFex, 2017-01-26:
I just finished setting up my Windows 7 64bit, system again, after a HD crash.

After installing 1.6.12 of the Arduino IDE, I am receiving this message:

`Invalid library found in C:\Users\USER\Documents\Arduino\libraries\FreeRTOS: C:\Users\kurtr.AVERNON\Documents\Arduino\libraries\FreeRTOS`

The FreeRTOS library version is: 9.0.0-1.

Any ideas what might be going on here?

feilipu, 2017-01-26:
If you've recovered your user files after the crash, then there's likely some left over directories etc from older Arduino versions. There were some substantial changes through to 1.8 (improvements).

I'd suggest following the directory trees indicated, and removing all files there under libraries. Note that there are two separate locations in your message.

Also, use [Arduino Version 1.8.1](https://www.arduino.cc/download_handler.php?f=/arduino-1.8.1-windows.exe). They've been fixing lots of stuff quickly, and I think (based on anecdotal evidence) that the later 1.6.x versions are not exactly bug free.

SalvorinFex, 2017-01-27:
HI, thanks for the info, that fixed it up.
Turns out they both were the same folder.

Closing.

closed 2017-01-27 by SalvorinFex