feilipu/Arduino_FreeRTOS_Library #13: Arduino Uno hangs beyond 60% code is flashed

dhamukrish, 2017-02-04:
Need your assistance why Arduino Uno hangs beyond 60% as I could not add even single line of code.

I have checked settings in the below files but could not figure out anything.
`FreeRTOSConfig.h`
`FreeRTOSVariant.h`

**Output**
```
Sketch uses 19,508 bytes (60%) of program storage space. Maximum is 32,256 bytes.
Global variables use 836 bytes (40%) of dynamic memory, leaving 1,212 bytes for local variables. Maximum is 2,048 bytes.
```

Would like to get your advise.

feilipu, 2017-02-05:
You are conflating several important memory concepts. The best thing to do is to understand the difference between "program storage space", aka Flash memory, and "Global Variables" and "Local Variables", aka RAM, in C and C++. Once you understand those differences then this answer will make more sense.

Your Uno is probably hanging because either the Idle Task stack space is being exceeded, or the stack in another Task is being exceeded. The Idle Task is equivalent to `loop()` when not using FreeRTOS.

You can check the usage of the stack assigned in FreeRTOS using the `uxTaskGetStackHighWaterMark(NULL)` function, as per [this example](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino).

You can add many more lines of code to the "Program Space". This has nothing to do with your Uno hanging.

dhamukrish, 2017-02-05:
Thanks @feilipu for your advise and I understand about memory concepts and am not sure whether I have explained properly.

Basically, I would like to re-iterate below are steps are successful even after adding the more lines within the program storage space.

**Successful steps**
1. Complied successfully with more lines
2. Sketch uploaded successfully to Uno

**Failed area of code**
I can see code is partially running as part setup() till task creation API and hangs. As none of my tasks are running beyond any additional/instructional within in existing task.

Please advise.

feilipu, 2017-02-05:
Good.

If you're having issues with task creation, then the best thing to do is to analyse what you are doing with the memory allocation for the each task. Note if your program cannot complete the task allocation there is a serious issue with your code.

If you are loading 60% of the Flash with your program, it is likely that it is already very large. Try breaking it down into smaller components, and try to get them working piece by piece.

Please read the FreeRTOS [task creation API](http://www.freertos.org/a00019.html) notes and [copy examples](https://github.com/feilipu/Arduino_FreeRTOS_Library/tree/master/examples) that are known to work, as a starting point.

If insufficient RAM remains following task creation to allow the creation of the Idle task, then there will be a fast LED blink to indicate the heap shortage, as noted in the [ReadMe](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/readme.md).

closed 2017-02-05 by feilipu

dhamukrish, 2017-02-05:
@feilipu Thanks for your advise and will modify the code as you described.