feilipu/Arduino_FreeRTOS_Library #14: More than one task is not create in arduino micro.

spoorthymb, 2017-02-08:
hi,
I am not able to create more than one task in arduino micro. i have used FreeRTOS example programs to start with the RTOS programming. 

#include <Arduino_FreeRTOS.h>

//

void setup()
{
  
  Serial.begin(9600);
  Serial.println(F("In Setup function"));

  /* Create two tasks with priorities 1 and 2. An idle task is also created, 
     which will be run when there are no tasks in RUN state */

 
 xTaskCreate(MyTask2, "Task2", 64, NULL, 1, NULL);
 xTaskCreate(MyTask1, "Task1", 64, NULL, 1, NULL);
 //xTaskCreate(MyIdleTask, "IdleTask", 100, NULL, 0, NULL);
  vTaskStartScheduler();
}

void loop()
{
  // DO nothing
}

/* Task1 with priority 1 */
static void MyTask1(void* pvParameters)
{
  while(1)
  {
    Serial.println(F("Task1"));
   //vTaskDelay(100/portTICK_PERIOD_MS);
   delay(200);
  }
}

/* Task2 with priority 2 */
static void MyTask2(void* pvParameters)
{
  while(1)
  {    
    Serial.println(F("Task2"));
    //vTaskDelay(150/portTICK_PERIOD_MS);  
    delay(100);
  }
}

/* Idle Task with priority Zero */ 
static void MyIdleTask(void* pvParameters)
{
  while(1)
  {
    Serial.println(F("Idle state"));
    delay(50);
  }

above was the program i had used.

feilipu, 2017-02-08:
You are not setting the Task Stack large enough. There is a configuration define for [`configMINIMAL_STACK_SIZE`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/FreeRTOSConfig.h) which is the smallest you can use.

Also, just for reference. The Idle Task is automatically defined by FreeRTOS, and in this library anything you put into the `loop()` function will be executed by the Idle Task. Task priority 0 is not the Idle Task.

spoorthymb, 2017-02-08:
what is your configTOTAL_HEAP_SIZE ? it is not mentioned in the configuration document. I have commented Idle task, its not been used i.

feilipu, 2017-02-08:
As noted in the [README](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/readme.md), the heap allocation is based on [`malloc()`](http://www.nongnu.org/avr-libc/user-manual/malloc.html) which takes its guidance from `__malloc_heap_start` and `__malloc_heap_end`. These values are provided for every AVR device automatically by avr-libc. When using `malloc()` there is no `configTOTAL_HEAP_SIZE`.

Also, please note that there no need to add `vTaskStartScheduler()` to your code. This function is already provided in the [`initVariant(void)`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/variantHooks.cpp) function. You won't find it inserted in the example code provided.

closed 2017-02-12 by feilipu

spoorthymb, 2017-02-14:
I was  using freeRtos v8.3 so it was not working. i replaced it by version 9, now it is working. thanks for your help.

Spoorthy