feilipu/Arduino_FreeRTOS_Library #15: Arduino IDE ver. 1.8.1 shows FreeRTOS as "INCOMPATIBLE"...

SalvorinFex, 2017-02-09:


feilipu, 2017-02-10:
Try to delete FreeRTOS out of your Arduino Libraries folder, then re-download it from the Library Manager. v9,0.0-1 is working fine, on my Win10 machine with IDE1.8.1
[image]

SalvorinFex, 2017-02-11:
OKay, I finally got back to this and now the library loads correctly.
So, as a test I loaded the Blink_AnalogRead.ino, but it seems like on the ProMicro, that the blink doesn't happen now. It just blinks once when I first connect to the Serial Monitor then when the numbers for the  AnalogRead come through, it doesn't blink.

feilipu, 2017-02-12:
I don't have a ProMicro (ATmega32U4) to test this issue, but if I did then I would try the following things.

- Revert the FreeRTOS library back to 8.2.3 and see if the issue is still apparent. If it is, then the issue will have been caused by some change in Arduino 1.8.1 that will be difficult to address.
- If the issue is not apparent with 8.2.3, but is with 9.0.0, then comment out the AnalogRead Task, and see what happens. If the Blink Task runs by itself, there may be a memory issue, somewhere.

Overall, I'm relatively sceptical that the ATmega32U4 with its USB interface is a good target for a simple FreeRTOS library. There are very tight timing constraints on the USB interface responses, and the Arduino platform "takes no prisoners" in dealing with these USB issues.

The ideal way to do a RTOS solution using the ATmega32u4 would be to integrate the RTOS into the kernel of the solution, rather than just a shim library in user space (to use Linux parlance), but this takes you outside the bounds of the Arduino experience.

SalvorinFex, 2017-02-13:
OKay, I followed the first procedure and it appears to be with the IDE 1.8.1.
I will be able to test 1.8.0 if that would help narrow it down?

I have found other items that also don't work in that newest version, so I will keep this in mind as well.

Thank you for your help...again.

feilipu, 2017-02-13:
No. Don't waste time in debugging Arduino. There are people doing that already.

Just find an IDE version that works for you, and be circumspect when upgrading.

closed 2017-02-13 by feilipu