feilipu/Arduino_FreeRTOS_Library #17: DS2482 not working

bkkrunal, 2017-04-16:
Hi Feilipu,
First, Thanks for your efforts for developing FreeRTOS.
I have been trying to read Temperature sensors DS18B20 via DS2482 IC which is TWI to One Wire Converter.
I can read the temperatures when the code is in loop, but when I insert the same code in task, the code freezes in that task.

Please find the code below:

For Loop function:
`
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Wire.h>

OneWire oneWire;
DallasTemperature sensors(&oneWire);
DeviceAddress currAddress;
int numberOfDevices=0;
float FinalTemp[15] = {0};

void setup()
{
  
  Serial.begin(19200);
  sensors.begin();
  
  Serial.println("Initialising...");

  // returns the number of devices(Temperature sensors) found on the bus 
  numberOfDevices = sensors.getDeviceCount();
  Serial.print("NoOfDevice= ");
  Serial.println(numberOfDevices);
  
  for (int i=0; i<numberOfDevices; i++)
  {
    // finds an address at a given index on the bus
    sensors.getAddress(currAddress, i);
    printAddress(currAddress);
    Serial.println();  
  }

  // sends command for all devices on the bus to perform a temperature conversion
    sensors.requestTemperatures();

     for(byte j=0; j<15; j++)
    {
      FinalTemp[j] = -999;
    }
    
    for (int i=0; i<numberOfDevices; i++)
    {
      Serial.print("Temp by Index");
      sensors.getAddress(currAddress, i);
      FinalTemp[i] = sensors.getTempCByIndex(i);
      Serial.print(FinalTemp[i]);
      Serial.println("°C");
      
      Serial.print("Temp by Address");
      Serial.print(sensors.getTempC(currAddress));
      Serial.println("°C");
    }
    
}

void loop() 
{
 Serial.println("Requesting temperatures...");
  sensors.requestTemperatures();
  uint8_t numberOfDevices = sensors.getDeviceCount();

  for(byte j=0; j<15; j++)
  {
    FinalTemp[j] = -999;
  }
  
  for (int i=0; i<numberOfDevices; i++)
  {
    sensors.getAddress(currAddress, i);
    printAddress(currAddress);
    Serial.print(": ");
    Serial.print(sensors.getTempCByIndex(i));
    FinalTemp[i] = sensors.getTempCByIndex(i);
    Serial.println();  
  }
  
}
`

`
For RTOS task

#include <Arduino_FreeRTOS.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Wire.h>

OneWire oneWire;
DallasTemperature sensors(&oneWire);
DeviceAddress currAddress;
int numberOfDevices=0;
float FinalTemp[15] = {0};

void setup()
{
  
  Serial.begin(19200);
  sensors.begin();
  
  Serial.println("Initialising...");

  // returns the number of devices(Temperature sensors) found on the bus 
  numberOfDevices = sensors.getDeviceCount();
  Serial.print("NoOfDevice= ");
  Serial.println(numberOfDevices);
  
  for (int i=0; i<numberOfDevices; i++)
  {
    // finds an address at a given index on the bus
    sensors.getAddress(currAddress, i);
    printAddress(currAddress);
    Serial.println();  
  }

  // sends command for all devices on the bus to perform a temperature conversion
    sensors.requestTemperatures();

     for(byte j=0; j<15; j++)
    {
      FinalTemp[j] = -999;
    }
    
    for (int i=0; i<numberOfDevices; i++)
    {
      Serial.print("Temp by Index");
      sensors.getAddress(currAddress, i);
      FinalTemp[i] = sensors.getTempCByIndex(i);
      Serial.print(FinalTemp[i]);
      Serial.println("°C");
      
      Serial.print("Temp by Address");
      Serial.print(sensors.getTempC(currAddress));
      Serial.println("°C");
    }

  xTaskCreate(
    SendDataTask
    ,  (const portCHAR *) "MonitorUART"
    ,  128  // Stack size
    ,  NULL
    ,  2  // Priority
    ,  &TaskHandle_SendData ); //Task Handle
    
  vTaskStartScheduler();    

}

void SendDataTask(void *pvParameters)  
{
  (void) pvParameters;
  for (;;) // A Task shall never return or exit.
  {  Serial.println("Requesting temperatures...");
  sensors.requestTemperatures();
  uint8_t numberOfDevices = sensors.getDeviceCount();

  for(byte j=0; j<15; j++)
  {
    FinalTemp[j] = -999;
  }
  
  for (int i=0; i<numberOfDevices; i++)
  {
      sensors.getAddress(currAddress, i);
      printAddress(currAddress);
      Serial.print(": ");
      Serial.print(sensors.getTempCByIndex(i));
      FinalTemp[i] = sensors.getTempCByIndex(i);
      Serial.println();  
  }
   }

}

`

feilipu, 2017-04-16:
I don't have any Dallas One Wire devices to debug this for you, but there are several things to note.

You don't need the `  vTaskStartScheduler();` function explicitly. It is included in the [initialisation code](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/variantHooks.cpp#L92). Just remove it.

You don't have an empty [`loop()`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/Blink_AnalogRead/Blink_AnalogRead.ino#L37) function in your code. Perhaps you've simply not declared it in the provided example. But I'm not sure what happens in the Arduino world if it is not provided. It may be doing something that we don't want to do.

I suggest using the `loop()` function to [put your Arduino into sleep mode](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L410), if nothing else.

You are running the SendDataTask without any pauses or delays. Do you really want the temperatures updated so quickly? Usually a call to [`vTaskDelay();`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/Blink_AnalogRead/Blink_AnalogRead.ino#L82) will ensure that you've got free processing time for other Tasks.

For debugging, always try to start with the Task with everything except a `Serial.print("OK");` and the stack highwater function commented out. That way you can be sure the Scheduler is correctly working as you expect.

And final point, the stack you've declared is only 128 bytes. Try increasing the stack size to something like 512 bytes. You can use the [stack highwater `uxTaskGetStackHighWaterMark()`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L298) function to check how full your stack is. But, if it starts out overflowed, nothing will happen. Also, don't cut the stack size too close (minimum 30 bytes), as some stack is used when the Task is swapped in and out.

bkkrunal, 2017-04-16:
Hi,
Thanks for fast reply.

Finally, I have increased the stack size, and now its working perfectly. 
Thanks a lot.

Also, Thanks for suggesting to use loop for putting Arduino into sleep. I will surely use this.

I do not need temperature to be updated so fast, I will add delays.
For adding delay, I have identified 3 functions:

1. Arduino's default:  delay(1000);
2. vTaskDelay();
3. vTaskDelayUntil();

Can you please put light on there differences in short.

feilipu, 2017-04-16:
Glad its working now.
Task stacks overflowing is the most common problem, I have to deal with.

All delay options will work.

1. Arduino's default: `delay(1000);` will busy wait the Task until it is swapped out. Typically you might prefer to do something else with the remaining time in a time slice, by blocking the Task (allowing the scheduler to swap it out), using one of below options, and swapping in another Task. Also, the delay time will not be accurate, because the `delay()` function will only run whilst the task is active. This option is NOT preferred.
2. `vTaskDelay();` will provide a delay based on the time slice provided by the scheduler (in this case the default is 15ms). So if you need to have 50ms delay, then you'll probably get 60ms as 4 slices is the closest result. This is the easiest option.
3. `vTaskDelayUntil();` using this function is good for precise amounts of long delay. If you need delays for accurate amounts of time (subject to the resolution of the time slice) then this is the best function, as it schedules the Task for unblocking at a specific System Tick count. To enable it you have to call the [ `xTaskGetTickCount();`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L211) function **ONCE** before your task goes into its infinite loop, to provide a local variable of [`TickType_t`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L206) type. This is the most useful option.

Enjoy.

closed 2017-04-16 by feilipu

bkkrunal, 2017-04-16:
Thanks, I got it now.
Thanks again a lot for all your detailed help.

Best Regards,
Krunal