feilipu/Arduino_FreeRTOS_Library #21: Calling a function inside a task

RwemaINYURU, 2017-06-30:
I created a task which reads GPS data but it has to call some function out of the task but when i am running them they are not returning anything.
What could be the cause of that?

feilipu, 2017-06-30:
I don't really know how to properly respond to the question, with so little information.

I suggest that there is nothing about FreeRTOS that would influence your issue. It is likely to be something to do with the way you're calling your function, or the way the data is being declared, before being returned.

To debug the problem, start by trying to flash a LED by calling a function from within the Task. Once that is working, then you're sure your function is being called.

You can then move on from there by returning a counter, or something else.
Use print statements to check where your code is executing.

Good luck.

RwemaINYURU, 2017-07-01:
I tried the code below to test it but it is printing `started` only the LED on 13 doesn't blink, what could be the cause? 
```c++ 
  #include <Arduino_FreeRTOS.h>

  void Blink(void *pvParameters);

  void setup() {
    Serial.begin(115200);

    xTaskCreate(Blink, (const portCHAR *)"blink", 100, NULL, 2, NULL);
  }

  void loop() {
    Serial.println("will loop infinitily");
  }

  void Blink(void *pvParameters) {
    int pin = 13;

    pinMode(pin, OUTPUT);

    while (true) {
      Serial.println("started");

      if (isPinHIGH == true) {
        digitalWrite(pin, LOW);
        delay(1000);
    }
    else {
      digitalWrite(pin, HIGH);
      delay(1000);
    }
  }
}

bool isPinHIGH() {
  if (digitalRead(13) == HIGH) {
    Serial.println("pin is HIGH");
    return true;
  }
  else {
    Serial.println("pin is LOW");
    return false;
  }
}

```

feilipu, 2017-07-01:
I'm 90% certain your issue is caused by not allocating enough Task stack. 

Using the serial.println() functions requires a lot of local memory.

Change the stack size to 500 bytes, and use the stack management tools to check the highwater levels 

Do not assume delay() will give you what you need. It is a busy wait, and relies on not being interrupted. See issue #20 for more.

RwemaINYURU, 2017-07-01:
I have removed the `delay()` replaced it with `vTaskDelay()` and added `vTaskDelay()` in the `void loop()` otherwise is printing this line `Serial.println("will loop infinitely");` infinitely, i left the stack memory to 100 and the led is blinking but it is not printing these lines `Serial.println("pin is HIGH");` and `Serial.println("pin is LOW");` in `bool isPinHIGH()` function even if i increase the stack memory to 10000. Below there is new code
```c++
#include <Arduino_FreeRTOS.h>

  void Blink(void *pvParameters);

  void setup() {
    Serial.begin(115200);

    xTaskCreate(Blink, (const portCHAR *)"blink", 100, NULL, 2, NULL);
  }

  void loop() {
    Serial.println("will loop infinitely");
    vTaskDelay(10000);  //Any number of ticks you enter it will work fine.
  }

  void Blink(void *pvParameters) {
    int pin = 13;

    pinMode(pin, OUTPUT);

    while (true) {
      Serial.println("started");

      if (isPinHIGH == true) {
        digitalWrite(pin, LOW);
        vTaskDelay(500);
    }
    else {
      digitalWrite(pin, HIGH);
      vTaskDelay(500);
    }
  }
}

bool isPinHIGH() {
  if (digitalRead(13) == HIGH) {
    Serial.println("pin is HIGH");
    return true;
  }
  else {
    Serial.println("pin is LOW");
    return false;
  }
}
```

feilipu, 2017-07-01:
Try removing the TaskDelay() from the loop(). I noted in the readme that the loop() shall never be blocked, because it is called from the IdleTask. Blocking loop() will kill everything.

Then go back to the example, and check it works. Change one thing at a time. See what breaks things. 

Use the stack high water management tools in FreeRTOS. Start with stack at 500 bytes, and reduce it from there.  That, and heap issues, are the cause of most pain.

C++ (Arduino) consumes lots of stack. That is the cost of convenience.

karawin, 2017-07-01:
isPinHIGH is a funtion which need ()
try:

#include <Arduino_FreeRTOS.h>
#define pin 13

  void Blink(void *pvParameters);

  void setup() {
    Serial.begin(115200);
    xTaskCreate(Blink, (const portCHAR *)"blink", 100, NULL, 2, NULL);
  }

  void loop() {
//    Serial.println("will loop infinitely");
//    vTaskDelay(10000);  //Any number of ticks you enter it will work fine.
  }

  void Blink(void *pvParameters) {    
    pinMode(pin, OUTPUT);
   Serial.println("started");
    while (true) {
      
      if (isPinHIGH() == true) {
        digitalWrite(pin, LOW);
      }
      else {
     	 digitalWrite(pin, HIGH);
      }
     vTaskDelay(500);
  }
}

bool isPinHIGH() {
  if (digitalRead(pin) == HIGH) {
    Serial.println("pin is HIGH");
    return true;
  }
  else {
    Serial.println("pin is LOW");
    return false;
  }
}

RwemaINYURU, 2017-07-01:
@feilipu if i comment out all lines in the `void loop()` an d I changed the stack to 500 now the led is not blinking any more only it is printing `started`.

@karawin i am using `isPinHIGH()` to try to simulate how to call a function inside a Task.

karawin, 2017-07-01:
ok but       
     if (isPinHIGH == true) {
is not correct.
      if (isPinHIGH() == true) {
is correct

RwemaINYURU, 2017-07-01:
Thank you @karawin i didn't noticed that now everything is working well.
But if we come back to the first comment i was trying to read GPS coordinates and I am using tinyGPS++ library this is the code I am using. 
```c++
#include <Arduino_FreeRTOS.h>
#include <TinyGPS++.h>

TinyGPSPlus gps;

void GPSreader(void *pvParameters);

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600);    //GPS module
  Serial.println("this is setup func");

  xTaskCreate(GPSreader, "GPSreader", 10000, NULL , 2, NULL);
}

void loop() {}

void GPSreader(void *pvParameters) {
  Serial.println("something atleast");
  while (true) {
    String values;
    char latitude[10], longitude[10], Altitude[8], Speed[10], satelites[3];

    printInt(gps.satellites.value(), gps.satellites.isValid(), 2, satelites);
    printFloat(gps.location.lat(), gps.location.isValid(), 9, 6, latitude);
    printFloat(gps.location.lng(), gps.location.isValid(), 9, 6, longitude);
    printFloat(gps.altitude.meters(), gps.altitude.isValid(), 7, 2, Altitude);
    printFloat(gps.speed.kmph(), gps.speed.isValid(), 6, 2, Speed);

    values  = latitude;
    values += ',';
    values += longitude;
    values += ',';
    values += Altitude;
    values += ',';
    values += Speed;
    Serial.print("GPS: ");
    Serial.println(values); //latitude, longitude, altitude, speed, direction
    //    vTaskDelay(100);
    smartDelay(100);
    if (millis() > 2000 && gps.charsProcessed() < 10)
      Serial.println(F("No GPS data received: check wiring"));
  }
}

void smartDelay(unsigned long ms) {
  unsigned long start = millis();
  do {
    while (Serial2.available())
      gps.encode(Serial2.read());
  } while (millis() - start < ms);
}

void printFloat(float val, bool valid, uint8_t len, uint8_t decPnt, char var[]) {
  if (!valid) {
    while (len-- > 1) {
      dtostrf(val, len, decPnt, var);
    }
  }
  else {
    dtostrf(val, len, decPnt, var);
  }
}

void printInt(uint8_t val, bool valid, uint8_t len, char var[]) {
  if (valid) {
    sprintf(var, "%d", val);
  }
  var[len] = 0;
  for (int i = strlen(var); i < len; ++i) {
    var[i] = '0';
  }
  if (len > 0) {
    var[len - 1] = ' ';
  }
}
``` 
these code works well without FreeRTOS. But now it is not starting the task, I know i used the `millis()` function and i was not supposed to use it so i am going to continue to find a way to remove it but anyone with an idea can help.
Thank you for the support.

karawin, 2017-07-01:
I think that you may use 
 xTaskGetTickCount
task.h
volatile TickType_t xTaskGetTickCount( void );
This function cannot be called from an ISR. Use xTaskGetTickCountFromISR() instead.
Returns:
    The count of ticks since vTaskStartScheduler was called.

feilipu, 2017-07-01:
You're asking for 10,000 bytes of stack, just for the GPSreader Task alone. If you're trying to run this on an Arduino Uno with 2048 bytes of RAM in total, then you've got a problem. The `xTaskCreate()` will certainly fail, and you'll get the `malloc()` fail LED blink (mentioned in the README) at 100ms.

Again, start with a Task stack of about 500 bytes. Not more, and use the [task highwater tools](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L267) to work out how much stack remains unused. Then reduce to leaving (say) 20 bytes free for each task.

Keep the `loop()` function empty, unless you use it to put the MCU to sleep when it is idle.

FreeRTOS has [xTaskDelayUntil()](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/examples/GoldilocksAnalogueTestSuite/GoldilocksAnalogueTestSuite.ino#L271) function that can delay a precise number of Ticks, usually divided by the Tick Period to get time in ms.

To set a one shot or multiple shot Timer, then use the FreeRTOS API for Timers. It is enabled in this port.

RwemaINYURU, 2017-07-01:
@feilipu i changed the stack to 500 bytes now it is fine but i am using arduino MEGA with 8Kb of RAM i don't understand why the i get `malloc()` fail LED blinking.
Thank you again.

feilipu, 2017-07-01:
Move these lines outside your `while(){ }` loop

```c++
    String values;
    char latitude[10], longitude[10], Altitude[8], Speed[10], satelites[3];
```
Whilst the compiler will probably optimise, and not keep on reallocating them, it is best not tempt fate.

And, `String` is in itself problematic to have on the stack. Arduino `String` functions are very RAM hungry. Try to move the `String values;` line to be a global (and therefore on the heap).

RwemaINYURU, 2017-07-01:
I moved those line above out of the `while(){}` and i made `String values;` variable global and I am getting `Serial.print(F(GPS: ));` printed but `Serial.println(values);` is printing nothing it seems that is not able to call `void printFloat(...)` and `void printInt(...)` functions or may be is because i am still using the `millis()`. About the serial port, for now i have one task thats why i didn't use the `semaphore`.

RwemaINYURU, 2017-07-04:
I have continued to try to solve that problem but i noted something that the `TinyGPS++` library am using is written in `C++` but `FreeRTOS` is in `C` and am trying to use `String` datatype which doesn't work with `C programming` so i also tried using `char*` of `C programming` which was able to return a string but i wasn't able to concatenate two variable with `char*` datatype. So any idea will be appreciated.

karawin, 2017-07-04:
strcat, strcpy, sprintf etc, learn a bit more about the possibilities.

RwemaINYURU, 2017-07-04:
Thanks a lot for everything now everything is working fine.
Below there is source code in case anyone may need them.
```c++
#include <Arduino_FreeRTOS.h>
#include <timers.h>
#include <TinyGPS++.h>

TinyGPSPlus gps;

void GPSreader(void *pvParameters);

TimerHandle_t GPStimer;
void GPSid() {}

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600);    //GPS module

  GPStimer = xTimerCreate("GPStimer", 100 / portTICK_PERIOD_MS, pdFALSE, GPSid, smartDelay);

  if (GPStimer == NULL)
    Serial.println(F("timer not created"));
  else
    xTimerStart(GPStimer, 200 / portTICK_PERIOD_MS);

  xTaskCreate(GPSreader, "GPSreader", 350, NULL , 2, NULL);
}

void GPSreader(void *pvParameters) {
  char* comma = ",";
  char values[70];

  while (true) {
    UBaseType_t uxHighWaterMark;
    Serial.println(uxTaskGetStackHighWaterMark(NULL));  //print unused stack memory

    char* Latitude = printFloats(gps.location.lat(), gps.location.isValid(), 9, 6);
    char* Longitude = printFloats(gps.location.lng(), gps.location.isValid(), 9, 6);
    char* Speed = printFloats(gps.speed.kmph(),   gps.speed.isValid(),    6, 2);
    char* SatNumb = printInts(gps.satellites.value(), gps.satellites.isValid(), 5);

    strcpy(values, Latitude);
    strcat(values, comma);
    strcat(values, Longitude);
    strcat(values, comma);
    strcat(values, Speed);

    Serial.println(values);

    vTaskDelay(200 / portTICK_PERIOD_MS);
    xTimerReset(GPStimer, 200 / portTICK_PERIOD_MS);
    if (xTaskGetTickCount() > 2000 && gps.charsProcessed() < 10)
      Serial.println(F("No GPS data received: check wiring"));
  }

}

void smartDelay(unsigned long ms) {
  while (Serial2.available())
    gps.encode(Serial2.read());
}
/********************************* NEW FOR GPS COORDINATES ******************************/
char* printFloats(float val, bool valid, int len, int decPnt) {
  char var[32];
  if (!valid) {
    while (len-- > 1) {
      var[15] = "********";
    }
  }
  else {
    dtostrf(val, len, decPnt, var);
  }
  return var;
}

char* printInts(unsigned long val, bool valid, int len) {
  char var[32];
  if (valid)
    sprintf(var, "%ld", val);
  var[len] = 0;
  for (int i = strlen(var); i < len; i++)
    var[i] = ' ';
  if (len > 0)
    var[len - 1] = ' ';
  return var;
}

void loop() {}
```

closed 2017-07-04 by RwemaINYURU