feilipu/Arduino_FreeRTOS_Library #22: Error compiling for Leonardo or Micro

cdrose, 2017-07-03:
I am trying to use Arduino FreeRTOS with an Arduino Micro (32U4), but I am getting a compile error with an empty sketch. Any ideas?

`
...\Arduino\sketchbook\libraries\FreeRTOS\src\variantHooks.c: In function 'initVariant':

...\Arduino\sketchbook\libraries\FreeRTOS\src\variantHooks.c:88:2: error: 'USBDevice' undeclared (first use in this function)

  USBDevice.attach();

  ^

...\Arduino\sketchbook\libraries\FreeRTOS\src\variantHooks.c:88:2: note: each undeclared identifier is reported only once for each function it appears in

exit status 1
Error compiling for board Arduino Leonardo.`

feilipu, 2017-07-03:
AFK. Will update this comment later.

USBattach() is the Arduino initialisation for USB devices, such as 32u4 and 32u2.

It may be that Arduino have changed the header or mechanism for initialisation, and now I'll need to update to reflect that.

Please note that because of the tight 1ms timing for USB keep alive messages the 32u4 devices can be problematic. Integration between USB interface and scheduler needs to be deeper than it is in this library.

feilipu, 2017-07-03:
Yes, Arduno have added setupUSB() to their initialisation, so I'll have to do a new release.

feilipu, 2017-07-03:
This code
```c++
#include <Arduino_FreeRTOS.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
Using Arduino v1.8.3 freshly installed, but 2nd compilation so that the core compilation is precompiled (cached).
```bash
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Phillip\AppData\Local\Arduino15\packages -hardware C:\Users\Phillip\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Phillip\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Phillip\Documents\Arduino\libraries -fqbn=arduino:avr:micro -ide-version=10803 -build-path C:\Users\Phillip\AppData\Local\Temp\arduino_build_77700 -warnings=all -build-cache C:\Users\Phillip\AppData\Local\Temp\arduino_cache_25044 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Phillip\AppData\Local\Temp\arduino_modified_sketch_923078\BareMinimum.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Phillip\AppData\Local\Arduino15\packages -hardware C:\Users\Phillip\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Phillip\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Phillip\Documents\Arduino\libraries -fqbn=arduino:avr:micro -ide-version=10803 -build-path C:\Users\Phillip\AppData\Local\Temp\arduino_build_77700 -warnings=all -build-cache C:\Users\Phillip\AppData\Local\Temp\arduino_cache_25044 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Phillip\AppData\Local\Temp\arduino_modified_sketch_923078\BareMinimum.ino
Using board 'micro' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\micro" "C:\Users\Phillip\AppData\Local\Temp\arduino_build_77700\sketch\BareMinimum.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\micro" "-IC:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src" "C:\Users\Phillip\AppData\Local\Temp\arduino_build_77700\sketch\BareMinimum.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\croutine.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\event_groups.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\heap_3.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\list.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\port.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\queue.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\tasks.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\timers.c
Using cached library dependencies for file: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS\src\variantHooks.cpp
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8037 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Micro"' "-IC:\Program Files (x86)\Ardui [... cut ...]
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -R .eeprom  "C:\Users\Phillip\AppData\Local\Temp\arduino_build_205864/BareMinimum.ino.elf" "C:\Users\Phillip\AppData\Local\Temp\arduino_build_205864/BareMinimum.ino.hex"
Using library FreeRTOS at version 9.0.0-1 in folder: C:\Users\Phillip\Documents\Arduino\libraries\FreeRTOS 
Sketch uses 10400 bytes (36%) of program storage space. Maximum is 28672 bytes.
Global variables use 300 bytes (11%) of dynamic memory, leaving 2260 bytes for local variables. Maximum is 2560 bytes.
```

cdrose, 2017-07-04:
I just asked for a library update and now it compiles fine for me too. No idea what happened there, but it seems to be working now. Thanks for looking into it!

feilipu, 2017-07-04:
Good to hear.

Please just note that because of the timing constraints on keeping the 32u4 (and 32u2) USB interface running (a 1ms cycle keep-alive message needs to be sent), and the fact that AVR has only one interrupt priority, this library's implementation of Scheduler timing via the WDT can run into issues.

In fact, sending a lot of data over the USB interface, combined with the keep-alive messages, can mean that the Scheduler interrupt is repeatedly missed, and therefore the wheels fall off everything.

The only way around this is to integrate the USB interface as a Task, or integrate the USB hardware timer with the Scheduler to ensure round robin handling. Both of which are outside the scope of this library solution.

closed 2017-07-04 by feilipu