feilipu/Arduino_FreeRTOS_Library #26: Compile time error

tiolatech, 2017-12-18:
I downloaded the library from eclipse. Upon first time compile,  I get the following error: 

`
C:\Users\timot\AppData\Local\Temp\ccD1BiXm.ltrans0.ltrans.o: In function `vPortYieldFromTick':
C:/Users/timot/.arduinocdt/libraries/FreeRTOS/10.0.0-3/src/port.c:478: undefined reference to `__EIND__'
C:/Users/timot/.arduinocdt/libraries/FreeRTOS/10.0.0-3/src/port.c:487: undefined reference to `__EIND__'
C:\Users\timot\AppData\Local\Temp\ccD1BiXm.ltrans0.ltrans.o: In function `vPortYield':
C:/Users/timot/.arduinocdt/libraries/FreeRTOS/10.0.0-3/src/port.c:461: undefined reference to `__EIND__'
C:/Users/timot/.arduinocdt/libraries/FreeRTOS/10.0.0-3/src/port.c:463: undefined reference to `__EIND__'
C:\Users\timot\AppData\Local\Temp\ccD1BiXm.ltrans0.ltrans.o: In function `xPortStartScheduler':
C:/Users/timot/.arduinocdt/libraries/FreeRTOS/10.0.0-3/src/port.c:434: undefined reference to `__EIND__'
`

tiolatech, 2017-12-19:
Error is the same when using the library when using it from the Arduino IDE as well

Floessie, 2017-12-21:
@tiolatech 
I updated this library in Arduino IDE 1.8.5 from 9.0 to 10.0.0-3 in the library manager (using the "Update" button) and the examples would no longer compile, though with a different undefined reference. I closed the IDE, removed the `libraries/FreeRTOS` folder and installed 10.0.0-3 anew. After that, the problems were gone.

Best,
Flössie

FrozenDroid, 2017-12-28:
Related to https://github.com/feilipu/avrfreertos/issues/5

feilipu, 2018-01-05:
Fixed in 10.0.0-7

closed 2018-01-05 by feilipu