feilipu/Arduino_FreeRTOS_Library #27: uxTaskGetStackHighWaterMark() return type too small

Floessie, 2017-12-21:
Hi Phillip,

I just stumbled upon the return type of `uxTaskGetStackHighWaterMark()`: It's `UBaseType_t`, and that is defined as [`unsigned char`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/portmacro.h#L64) in this port. The stack size is provided as `unsigned short` words to [xTaskCreate()](https://www.freertos.org/a00125.html), thus theoretically 64kB are possible here.

Wouldn't it be better to patch [`tasks.c`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/tasks.c#L3645) to make it return `unsigned short` instead of `UBaseType_t`, as [`prvTaskCheckFreeStackSpace()`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/tasks.c#L3625) correctly returns a `uint16_t`? Otherwise `uxTaskGetStackHighWaterMark()` will return a misguiding value.

Thanks,
Flössie

feilipu, 2018-01-01:
@Floessie,

that seems correct.

In the current v10 version the stack size type is configurable [`configSTACK_DEPTH_TYPE`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/task.h#L234) and is defined [here as `uint16t`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/Arduino_FreeRTOS.h#L828). So the documentation is slightly out of date.

I think that @RichardBarry is still in the process of sorting out a migration in this direction, and so it is something that should interest him.

I'll leave this open and see what happens.

closed 2018-01-03 by feilipu

feilipu, 2018-01-03:
I found one more reference, and added the backwards compatibility option discussed.

(also reverted the __EIND__ references. still can't understand why the #definition was not being expanded in the macro. anyway... whatever).

RichardBarry, 2018-01-03:
Not been following the EIND discussion recently, but from your last comment "still can't understand why the #definition was not being expanded in the macro" I'm surmising the EIND was being #defined to something that was not being pre processed as expected.  If that is the case, and if the macro was in an assembler file with a .s extension, then try changing the extension to .S (upper case) if it is to be preprocessed.

feilipu, 2018-01-03:
Off issue topic, but I was trying to prettify the [`portSAVE_CONTEXT` macro](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/port.c#L91) by using `__EIND__` and a define for this in `portmacro.h`. So, there were no .s or .S files involved.

Anyway, it was just to make things pretty, so I reverted it to use the `0x3C` value.

PR #29 (2018-01-03): Fix too small return type for `uxTaskGetStackHighWaterMark()`
This PR fixes #27 by replacing the `uint8_t` return type of `uxTaskGetStackHighWaterMark()` with `uint16_t`.