feilipu/Arduino_FreeRTOS_Library #28: What's the correct way to yield from ISR here?

Floessie, 2017-12-21:
Hello Phillip,

The [`*GiveFromISR()`](https://www.freertos.org/a00124.html) functions have a `pxHigherPriorityTaskWoken` parameter. This can be optionally NULL, but the examples make use of it, stating *the actual macro used here is port specific*.

What's the right macro to use in this AVR port from your point of view? And if it's a macro (and not a function), can it be handled port agnostic using some `#ifdef`s?

Thanks,
Flössie

feilipu, 2018-01-01:
I think that a task context switch with a `reti` instruction terminating is required.
I'll have to look into this at some stage. I don't have an immediate answer.

Floessie, 2018-01-01:
Hi Phillip,

Thanks for looking into it!

Best,
Flössie

Floessie, 2018-01-04:
So this basically means, we can use `vPortYield()` here?

Floessie, 2018-01-04:
Yet another [hint](https://exploreembedded.com/wiki/Resuming_Task_From_ISR) into that direction.

feilipu, 2018-01-04:
I think the right answer is to know the preamble for entering an ISR, and then reverse it.

Since we're calling the ISR with `__attribute__ ((naked))` there is no avr-libc preamble to worry about.
We just have to unroll our own preamble. That isn't calling `vPortYield()` since that will roll another context save.

I think (untested) that we just need to do the context switch, and then remember to unroll the ISR context save with a restore. The issue is how the restore context happens. It needs to be before the context switch, or the ISR will never get reenabled, and we're stuck in it forever.

But I'm not sure how to ensure the context switch happens after the ISR context restore. Perhaps there needs to be an ISR context restore, and then a "jump" to the `vPortYield()` routine, where the context can be saved without masking the ISR return address.

I know how to engineer an assembly "jump" in Z80 C code, but I'm not sure how how to do it from C in avr assembly.

Floessie, 2018-01-08:
Any outcome? I searched the Internet once more but found nothing new...

feilipu, 2018-01-09:
Need some time to think, code, and test, which is lacking currently.

The thing I’m most conserned about is the return address pushed by the ISR when it is entered, and where it has to end up if the ISR exits to a different address.

Floessie, 2018-01-09:
Coincidentally, I just found [this explanation](https://www.freertos.org/FreeRTOS_Support_Forum_Archive/June_2005/freertos_taskYIELD_in_interrupt_1299296.html) from Richard, dating back to 2005 but nevertheless. Maybe it's helpful. (I mean the long explanation somewhere near the end of the page.)

Best,
Flössie

feilipu, 2018-01-09:
Richard’s 14 June comment is detailed. And comprehensive. But, I don’t fully understand it yet.

Will need to draw some stacks to know what to do, and it will take some time.

feilipu, 2018-01-09:
But the fact program returns to within the ISR makes handling the stacked return address easy.

feilipu, 2018-01-09:
This is the sequence provided by Richard, and edited by me for reference.

1) Task A is running when an interrupt occurs.  The interrupt  causes some CPU registers to be saved including the interrupt flags.  Registers are saved so as to be restored, _and a return address is pushed by C which is popped back to the PC_ by a RETI instruction.  For the interrupt to have occurred the interrupt flags must have been set with interrupts enabled - but within the ISR they are disabled.

2) The ISR code executes.  During the execution it is discovered that a yield is required.  The yield does not occur however - just  a flag is set to say one is necessary.

3) At the end of the interrupt the flag is checked which says a yield is required, and yield is called.

4) Yield saves the context of the task A - which has been interrupted.  The registers are saved to the stack on top of the registers that were already pushed onto the stack when the ISR was entered.  The program counter that is saved points to the end of the ISR function. _So there are stacked PCs stored now. And stacked registers._

5) Yield then switches the stack pointer to point to the saved context of task B - and proceeds to pop the registers from task B context.  When all the general purpose registers have been popped the task B interrupt flags and return address are left on the stack.  The flags are popped first.  This will return the interrupt enable status to whatever it was when task B last ran (could be enabled or disabled).  Finally RET is executed which causes the return address to be popped and the program counter set to the correct place to run task B.

6) At some point task B blocks.  In our example assume this makes task A the candidate to run again.  Having saved the task B context the context switch this time sets the stack pointer back to the saved context of task A (_no the ISR status of the task A, as that is top of stack_) - the task that was interrupted. 

7) The general purpose registers for task A (_in the ISR_) are restored.  Next the interrupt flags are restored.  The interrupt flags were stored with interrupts disabled as they were saved from within the ISR.  Finally the RET instruction is executed which sets the program counter back to the place where task A was last executing.  In this case task A was last executing in the ISR.  The next items on the stack are the registers pushed onto the stack by the processor when the interrupt occurred. [NB there may actually be some registers on the stack from the function prologue generated by the compiler, which will be popped by the epilogue generated by the compiler].

8) Task A returns to within the ISR.  The ISR processing has already been performed but the ISR itself has not finished.  The next instruction that task A executes is from the function epilogue - in a RETI.  This pops the interrupt flags and return address from the stack - these are those placed there when the ISR was entered.  The interrupt flags re-enables interrupts (remember they must have been enabled for the interrupt to get called at all) and returns task A to the place it was prior to it getting interrupt in the first place.

Floessie, 2018-01-10:
I think, your edition above is correct. So I conclude the following:

- `taskYield()` can be used to yield from an ISR today. Just as the earlier hints told us. From the above I also conclude, that this `taskYield()` doesn't necessarily have to be the last call in the ISR, as the ISR context is completely restored (as yet).
- Your are thinking about an optimized solution, that wouldn't restore the ISR context just to restore task A context. This would require the new `portYIELD_FROM_ISR()` to be the very last call in the ISR, enforcing an `_attribute__((always_inline))` for a function wrapping this call (which is okay).
- What about those registers on stack generated by the prologue mentioned in 7.? Can they be ignored?
- When the SREG in 1. is saved, the interrupt flag is *not* set, isn't it? So when you manipulate the stack for your optimized return, you'll have to modify the stored SREG, too, as the task switch will use a RET instead of a RETI. Your last paragraph was a bit puzzling in that regard...

Bottom line: Until those optimizations are in place, one can safely use `taskYield()` to yield from an ISR in this very port.

Thanks,
Flössie

feilipu, 2018-01-10:
> What about those registers on stack generated by the prologue mentioned in 7.? Can they be ignored?

As the ISR created with the `__attribute__((naked))` decoration, the only register that would be pushed is the PC at the time when the ISR interrupted normal program flow. And, this is the very register that we need to avoid bouncing back.

But, still we need to have a clear think about this, and check the asm code generated.

JoshuaKim6502, 2019-09-03:
Hi everyone,
If the parameter “xHigherPriorityTaskWoken” in the function  “*GiveFromISR()” results “pdTRUE,” it means that the “*GiveFromISR()” causes a task to leave the Blocked state, and the unblocked task has a higher priority than the interrupted task.   In this case, ending the ISR with taskYIELD() function let the unblocked task run immediately after the ISR. Otherwise, the interrupted task resumes after the ISR, and then the unblocked task runs.
Thanks a lot for Phillip’s latest update version 10.2.0-3, with which I have tested.

feilipu, 2020-04-11:
I think this is now done.

closed 2020-04-11 by feilipu

Floessie, 2020-04-16:
I'll adapt to `portYIELD_FROM_ISR()` ASAP. :+1:

feilipu, 2020-04-16:
I should do one more alignment release, now that [ATmegaxxxx](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/48) has been merged. Still need to get `portYIELD_FROM_ISR()` into a release.

Floessie, 2020-04-16:
As `portYIELD_FROM_ISR` is a macro, I think I'll `#ifdef` it, so it doesn't matter if it's already present or not...

Thanks for you efforts! :+1: The new AVR128[DA](https://www.microchip.com/wwwproducts/en/AVR128DA28)/[DB](https://www.microchip.com/wwwproducts/en/AVR128DB28) series will be a perfect fit for FreeRTOS. Looking forward to Arduino and FreeRTOS support for it, which will surely need some time to mature.

Best,
Flössie

feilipu, 2020-04-16:
The [AVR128DA28](https://www.microchip.com/wwwproducts/en/AVR128DA28)/[AVR128DB28](https://www.microchip.com/wwwproducts/en/AVR128DB28) is like someone put most of the [Goldilocks Analogue](https://www.kickstarter.com/projects/feilipu/goldilocks-analogue-classic-arduino-audio-superpow) features in one chip. Finally.

But, the integrated DAC is a bit weak, given what they could easily have done by integrating the [MCP](https://www.microchip.com/wwwproducts/en/MCP4822) product range, now its even part of the same company.