feilipu/Arduino_FreeRTOS_Library #30: undefined reference while using co-routines.

Meresmata, 2017-12-23:
Hi, for didactic reason, I want write a sketch with co-routines and as well with tasks...

Today I tried the one with co-routines. And resulted with the following code: 
```

#include <LiquidCrystal.h>

#include <Arduino_FreeRTOS.h>
#include <croutine.h>

//define coroutine

void vApplicationIdleHook();
void vaCoDisplayRoutine(CoRoutineHandle_t xHandle, UBaseType_t uxHandle);

//globales
 LiquidCrystal lcd(12, 11, 10, 5, 4, 3, 2);

 
//the setup function runs once when you press reset or power the board
void setup() {
 
      TaskHandle_t xHandle;
      UBaseType_t uxIndex;
     // Create two co-routines at priority 1.
     for( uxIndex = 0; uxIndex < 2; uxIndex++ )
     {
         xCoRoutineCreate( vaCoDisplayRoutine, 1, uxIndex);
     } 

     vApplicationIdleHook();
}

void loop()
{
  // Empty. Things are done in Tasks.
}

void vaCoDisplayRoutine(CoRoutineHandle_t xHandle, UBaseType_t uxHandle){
  static const uint32_t xDelayTime = 200 / portTICK_PERIOD_MS;
  
  crSTART(xHandle);
  lcd.begin(16,1);
  for(;;)
    lcd.print("Co-routine " + uxHandle);
    crDELAY( xHandle, xDelayTime);
  ;
  crEND();
}

void vApplicationIdleHook()
{
        vCoRoutineSchedule();
}

```

When I try to compile it I get the following error messages while linking:

> /tmp/cc6d60V8.ltrans0.ltrans.o: In function `vApplicationIdleHook':
> FreeRTOS.ino:49: undefined reference to `vCoRoutineSchedule'
> /tmp/cc6d60V8.ltrans0.ltrans.o: In function `setup':
> /FreeRTOS.ino:24: undefined reference to `xCoRoutineCreate'
> FreeRTOS/FreeRTOS.ino:24: undefined reference to `xCoRoutineCreate'
> /tmp/cc6d60V8.ltrans0.ltrans.o: In function `vApplicationIdleHook':
> FreeRTOS.ino:49: undefined reference to `vCoRoutineSchedule'
> collect2: error: ld returned 1 exit status

Does anyone know why? 

By the way: the type xTickType is not defined (found here: https://www.freertos.org/crdelay.html).
Shall I write a highlighting scheme for the Arduino IDE?

Floessie, 2017-12-29:
@Meresmata 
> Does anyone know why?

Because coroutines [aren't configured](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/FreeRTOSConfig.h#L77)? Try to replace that `0` with a `1` in your copy.

HTH,
Flössie

Meresmata, 2017-12-30:
Hi,

 that was the best and most precise help I have ever got. 

Unfortionately, there is still a problem with in the co-routines.`#include <Arduino_FreeRTOS.h>
#include <LiquidCrystal.h>
#include <croutine.h>

//define coroutine

void vApplicationIdleHook();
void vaCoDisplayRoutine(CoRoutineHandle_t xHandle, UBaseType_t uxHandle);

//globales
 LiquidCrystal lcd(7, 8, 9, 10, 11, 12);

 
//the setup function runs once when you press reset or power the board
void setup() {
 
      //TaskHandle_t xHandle;
      UBaseType_t uxIndex;
     // Create two co-routines at priority 2.
     for( uxIndex = 0; uxIndex < 1; uxIndex++ )
     {
         xCoRoutineCreate( vaCoDisplayRoutine, 2, uxIndex);
     } 

     vApplicationIdleHook();
}

void loop()
{
  // Empty. Things are done in Tasks.
}

void vaCoDisplayRoutine(CoRoutineHandle_t xHandle, UBaseType_t uxHandle){
  static const uint32_t xDelayTime = 1000 / portTICK_PERIOD_MS;
  
  crSTART(xHandle);
  lcd.begin(16,1);
  for(;;) {
    lcd.clear();
    lcd.setCursor(0,1);
    lcd.print("Co-routine " + String(uxHandle));
    crDELAY( xHandle, xDelayTime);
  }
  crEND();
  
}

void vApplicationIdleHook()
{
        vCoRoutineSchedule();`
}`

`

Because, after the end of the delay time, the display got black. I also have a problem, when to instances  of the co-routines use the display... but I guess, that can be solved with the semaphores?

All also write a version with tasks, but that one does not write anything to the display at all. Testing help me to find out, that it has something to do with the line:
`    lcd.print("Task " + String(*(uint16_t*)pvParameters));`

Full code is: `#include <Arduino_FreeRTOS.h>
#include <LiquidCrystal.h>
#include <stdint.h>

void vApplicationIdleHook();

//define tasks
void vTaskDisplayRoutine(void *pvParameters);

//globales
 LiquidCrystal lcd(7, 8, 9, 10, 11, 12);

 
//the setup function runs once when you press reset or power the board
void setup() {
 
      uint16_t pVParam;
     // Create two tasks at priority 2.
     for( pVParam = 0; pVParam < 1; pVParam++ )
     {
        xTaskCreate(
          vTaskDisplayRoutine,
          (const char * const)("TaskBlink"/*+String(pVParam)*/),
          128,
          (void *) pVParam,
          2,
          NULL
          );
     } 

     vApplicationIdleHook();
}

void loop()
{
  // Empty. Things are done in Tasks.
}

void vTaskDisplayRoutine(void *pvParameters){
  static const uint32_t xDelayTime = 1000 / portTICK_PERIOD_MS;

  lcd.begin(16,1);
  for(;;) {
    lcd.clear();
    lcd.setCursor(0,1);
    lcd.print("Task " + String(*(uint16_t*)pvParameters));
    vTaskDelay(xDelayTime);
  }

  
}
`

Perhaps you can help me out once again and. Perhaps I to write the code totally different?

feilipu, 2018-01-01:
If you are planning on using co-routines, then I'm afraid that I can't comment. I haven't used this feature previously.

There is a lot of information on the web site on [Tasks and Co-routines](https://www.freertos.org/taskandcr.html), and further [advanced information](https://www.freertos.org/croutine.html).

But, I assume you've read and digested all of that information already. And, have tried the [demonstration examples](https://www.freertos.org/coroutine-example-code.html) provided.

If you get it sorted, please leave a comment here. Good luck.

closed 2018-01-09 by feilipu