feilipu/Arduino_FreeRTOS_Library #31: SD card functions not working.

bkkrunal, 2017-12-30:
Hi Phillip,

I am trying to get SD card working with the FreeRTOS, but it does not work at all. 
I even disabled interrupts but with no luck. 
The SD functions always return 0.
When I try this in loop() function by disabling #include <Arduino_FreeRTOS.h>, it works perfect.
Please suggest,

noInterrupts(); 
    myFile = SD.open("171230s.csv", FILE_WRITE);
    Serial.println(myFile);    //Returns 0
    Serial.println(myFile.println("1,2,3,4,5,6"));    //Returns 0
    myFile.close();
    interrupts();

feilipu, 2018-01-01:
The first issue you have is that the `loop()` function executed by the FreeRTOS idle task, and you are disabling interrupts and blocking in the idle task. Remember that the `loop()` function should never disable interrupts or block processing.

Also, the idle task has a small stack allocated of 192 bytes. Beyond very simple examples this will not be sufficient, and is certainly not enough for implementing `println()`.

Any real things should be done in tasks (as per both the simple examples provided), and `loop()` should be left empty of everything except setting the MCU into idle or sleep.

bkkrunal, 2018-01-01:
Hi, Thanks for the reply.
Sorry for not being clear on the issue.

By loop function, I meant completely removing FreeRTOS and going by the normal approach.
So I have tested 2 options, 1 by using FreeRTOS and other without it.

I used the same following function for both the options,

`noInterrupts();
myFile = SD.open("171230s.csv", FILE_WRITE);
Serial.println(myFile); //Returns 0
Serial.println(myFile.println("1,2,3,4,5,6")); //Returns 0
myFile.close();
interrupts();`

I actually wanted to point out that, this worked well when used without RTOS and doesn't work with RTOS. I hope I am clear now.

feilipu, 2018-01-01:
Similarly, the first issue have is that  you are disabling interrupts for potentially several hundred milliseconds and therefore critically disabling the RTOS scheduler.

Any code using Arduino (or other) SD card functions has to be aware that SD cards have hundreds of milli-seconds of delay built into their interfaces. You can't use an SD card with the global interrupt disabled and expect that a RTOS can survive.

Why are you disabling interrupts in this code?

bkkrunal, 2018-01-01:
Actually, I tried it previously without disabling interrupts, but it was not working. 
I thought it would give all the CPU resources to SD card routine only if I disable the interrupts.
But, it does not work both the way.

bkkrunal, 2018-01-02:
Hi, I tried with different SD card library and now it is working as expected.
This is the new SD library.
https://github.com/greiman/SdFat
So it seems to be SD lib problem and not with the FreeRTOS.
Thank you very much for your help till now.

Best Regards,
Krunal

feilipu, 2018-01-02:
Good to hear.
I've also got a working version, using [ChaN's FatFS](https://github.com/feilipu/avrfreertos/tree/master/freeRTOS10xx/lib_fatf) library.
But note that it is in no way Arduino compatible.

closed 2018-01-02 by feilipu