feilipu/Arduino_FreeRTOS_Library #34: Use other timer than WDT in arduino_FreeRTOS_Library

SigP226, 2018-02-08:
Hello,

my configuration:
Eclipse Oxygen 2 (on windows 7)
Avr plugin
arduino source code
Arduino_FreeRTOS_Library (10.x)
a simple "blink" program
board : meduino2560 

Everything compile and work fine but I would like to use an other timer than watchdog timer to reduce tick to 1ms. I tried to understand how to extract interesting code from AvrFreeRTOS (freeRTOS10xx repository) which have this timer configuration, but I resign, don't have enough skill.

Is there any "simple" way to add the timer 1/2/3 configuration for tick in the Arduino_FreeRTOS_Library?
which file do I need to modify?

Eric

SigP226, 2018-02-09:
Edit
seems solution found (thank Eclispe file comparator)
use file port.c fromAvrFreeRTOS/freeRTOS10xx
use some macro from FreeRTOSboarddefs.h and transfer it in FreeRTOSVariant.h
modify portmacro.h

Still testing but seem to work 

Eric

Floessie, 2018-02-18:
@SigP226 Eric, I think, that's a good idea. Using Timer3 if available could make the library even more useful. Do you have any code to share? You can easily fork this library and create a branch with enhancements.

Best,
Flössie

feilipu, 2018-02-20:
I have a `port.c` file for every timer config in the `AVRfreeRTOS` repository. In particular, ATmega328p, ATmega1284p, and ATmega2560 are supported.

But of course using the other timers breaks the generality of the library because each ATmega MCU has different timer capabilities, and the Arduino core configures all the normal timers (whether it uses them or not), so the standard core will be broken.

If you want to dig deeper, then you’re better off ditching Arduino and just using the parts you need for your particular solution.

closed 2018-02-21 by feilipu

anatom74, 2018-03-13:
> I have a port.c file for every timer config in the AVRfreeRTOS repository. In particular, ATmega328p, ATmega1284p, and ATmega2560 are supported.

Where could I find those files. I would like to support mega1280 with Timer3

anatom74, 2018-09-08:
using port.c from [AVRFreeRTOS](https://github.com/feilipu/avrfreertos) repo and modifications as proposed by SigP226, I've been able to successfully run FreeRTOS on Timer3, but I don't have accurate timing at all.
using vTaskDelay(200) gives me actual delay of ~430ms instead of expected 1000ms.

Anyone already had this issue?

feilipu, 2018-09-08:
@anatom74 what you're probably finding is the issue with Arduino, that it possesses and configures ALL the hardware available on the ATmega device.

So, despite the fact that [you think you're configuring the timer to work correctly](https://github.com/feilipu/avrfreertos/blob/master/freeRTOS10xx/portable/port.c#L654), because you're using the correct initialisation routine from my code, Arduino is coming in later and [reconfiguring ALL](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring.c#L241) the timers (here [specifically Timer 3](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring.c#L326)) to suit what it needs. Mine,... its all mine.

There really isn't a clean way to work with Arduino and be deterministic about timing and memory, and keep hardware (any ATmega MCU) generality. But, if you don't use Arduino then you can't rely on its benefits, either. `ROCK>|<HARD_PLACE`.