feilipu/Arduino_FreeRTOS_Library #35: Help needed, cannot make ethernet and Serial work together.

bkkrunal, 2018-02-18:
Hi Phillip,
I am trying to use hardware usart and ethernet shield V2 (having W5500) together on MEGA2560.
I first initialise ethernet module in setup(), which works fine.
Then I initialize serial port with RTOS task and try to read the data from port using this task.
The problem I am facing is that I never receive any data on serial port after initialising the ethernet.
However, I am able to send data using serial port even after ethernet initialisation. 
I observed that my Serial.available() in the task never returns any value other than 0.
If I just remove the ethernet initialising code, then I can receive and send data on the serial port.
I am using this library for ethernet shield.
https://github.com/PaulStoffregen/Ethernet

I tried by increasing the stack size of the task as well with no luck.
Please suggest.

Regards,
BK Krunal

`
#include <SPI.h>
#include <Ethernet.h>
#include <Arduino_FreeRTOS.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192, 168, 43, 177);
char temp;

void setup() 
{
  Serial.begin(115200);
  Serial1.begin(115200);

 pinMode(25, OUTPUT);   //Relay Driver_CS
 digitalWrite(25, HIGH);
  
 pinMode(A4, OUTPUT);  // SD_CS
 digitalWrite(A4, HIGH);
  
 pinMode(53, OUTPUT);  // Ethernet CS
 digitalWrite(53, HIGH);
   
 pinMode(9, OUTPUT); //Reset pin of W5500
 digitalWrite(9, HIGH);
 
 Serial.println(F("Initialising Ethernet..."));
 if (Ethernet.begin(mac) == 0)
 {
    Serial.println(F("Failed to configure Ethernet using DHCP"));
    Ethernet.begin(mac, ip);

      Serial.print(F("IP= "));
      Serial.println(Ethernet.localIP());
 }
 else
 {
      Serial.print(F("IP= "));
      Serial.println(Ethernet.localIP());
 }

 xTaskCreate(
    DebugMonitor
    ,  (const portCHAR *) "MonitorUSART"
    ,  128  // Stack size
    ,  NULL
    ,  2  // Priority
    ,  NULL ); //Task Handle

}

void DebugMonitor(void *pvParameters)  
{  
  (void) pvParameters;
  
  for (;;) // A Task shall never return or exit.
  { 

    while(Serial.available() > 0)
    {
        temp = Serial.read(); 
        Serial1.write(temp);
    }
  }
}

void loop()
{
  
}
`

Floessie, 2018-02-18:
@bkkrunal I can't comment on your W5500 problem, but your `DebugMonitor` function loops madly without ever yielding to another task. How about sleeping at least one tick, if there's nothing to read?

You should reformat your code above like so
\```c++
// Code
\```

HTH,
Flössie

bkkrunal, 2018-02-18:
Thanks for the reply.
Can you please suggest how I can do it? I would like to give it a try.
I will get the code for sleeping Arduino, I just need to understand where should I put it.

bkkrunal, 2018-02-18:
Hi,
I have got this link for the reference.
https://feilipu.me/2015/11/24/arduino_freertos/
This has explained how to do it.
Thanks for your replies.

Floessie, 2018-02-18:
@bkkrunal 
> Can you please suggest how I can do it? I would like to give it a try.

Something like this:

```c++
void DebugMonitor(void *pvParameters)  
{  
  (void) pvParameters;
  
  for (;;) // A Task shall never return or exit.
  { 

    while(Serial.available() > 0)
    {
        temp = Serial.read(); 
        Serial1.write(temp);
    }

    vTaskDelay(1); // Nothing was available, give other tasks a chance
  }
}
```

HTH,
Flössie

bkkrunal, 2018-02-18:
Thanks for the reply.
But that didn't make any change.
Still, do not receive any data on serial port.

CunningLearner, 2018-09-03:
Hi,
I have tried the same with ethernet shield W5100 on MEGA2560 and its working even after ethernet initialization. I am using serial terminal to get the data entered from Serial(reading from Serial terminal) to be sent on Serial1. Also, I am using the same ethernet library and is able to recieve the data from Serial1 of MEGA on Arduino nano as well as USB to TTL converter before and after initialization of ethernet.
```c++
#include <SPI.h>
#include <Ethernet.h>
#include <Arduino_FreeRTOS.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192, 168, 43, 177);
char temp;

void setup() 
{
  Serial.begin(115200);
  Serial1.begin(115200);

 pinMode(25, OUTPUT);   //Relay Driver_CS
 digitalWrite(25, HIGH);
  
 pinMode(A4, OUTPUT);  // SD_CS
 digitalWrite(A4, HIGH);
  
 pinMode(53, OUTPUT);  // Ethernet CS
 digitalWrite(53, HIGH);
   
 pinMode(9, OUTPUT); //Reset pin of W5100
 digitalWrite(9, HIGH);
 
 Serial.println(F("Initialising Ethernet..."));
 if (Ethernet.begin(mac) == 0)
 {
    Serial.println(F("Failed to configure Ethernet using DHCP"));
    Ethernet.begin(mac, ip);

      Serial.print(F("IP= "));
      Serial.println(Ethernet.localIP());
 }
 else
 {
      Serial.print(F("IP= "));
      Serial.println(Ethernet.localIP());
 }

 xTaskCreate(
    DebugMonitor
    ,  (const portCHAR *) "MonitorUSART"
    ,  128  // Stack size
    ,  NULL
    ,  2  // Priority
    ,  NULL ); //Task Handle

}

void DebugMonitor(void *pvParameters)  
{  
  (void) pvParameters;

  for (;;) // A Task shall never return or exit.
  { 
    while(Serial.available() > 0)
    {
        temp = Serial.read(); 
        //Serial.println(temp); //Checked--> the data recieved from Serial
        Serial1.write(temp);    //Sending the recieved data to Serial1
    }
    vTaskDelay(1);
  }
}

void loop()
{
  
}
```
Also, this is the code I am using to receive data from Serial1 of MEGA to Arduino nano:
```c++
char temp;
void setup() {
  // Setting up the baud for synchronization
Serial.begin(115200);
}

void loop() {
  // Recieving data if available.
 while(Serial.available() > 0)
    {
        temp = Serial.read(); 
        Serial.println(temp);
    }
}
```
I am not sure about ethernet W5500 as I haven't tried on the same. Hopefully, it is the issue with the ethernet and not with the Arduino_FreeRTOS compatibility of making the Serial and ethernet work together.

closed 2018-09-05 by feilipu