feilipu/Arduino_FreeRTOS_Library #37: Watchdog reset

anatom74, 2018-03-13:
Hi,
What would be the easieast way to reset MCU (Watchdog reset)?
I would need this so I can jump to bootloader from app.

Regards,
A

feilipu, 2018-03-14:
The best way to do what you want is not with the watchdog timer.

There is a special ATmega function to do what you require. Find the datasheet relevant for the device you're using, and search for "Boot Loader Support – Read-While-Write Self-Programming". There you will find information on the Fuse which sets the Reset Vector, which is used for jumping to the bootloader from an app.

Depending on where your bootloader is located (which is determined by its size),  the bootloader address could be `0x1C00`, `0x1E00`, or another address configured by the Boot Size Configuration bits. You will need to jump there from your app to initiate your bootloader code.

closed 2018-03-17 by feilipu

anatom74, 2018-06-22:
Still there is a problem if not doing system reset as some registers have old values. The cleanest is to do it with Watchdog System Reset.

Would it be possible to do it like:
- stopping scheduler
- reconfiguring watchdog to System Reset
- go into infinite loop without resetting WDT?

feilipu, 2018-06-24:
I don't think you need to stop the scheduler. If you're doing a reset, then you're free from that responsibility.

There is an AVR library [`#include <avr/wdt.h>`](https://www.nongnu.org/avr-libc/user-manual/wdt_8h_source.html) with the `wdt_enable()` function required to reconfigure the watchdog timer to reset, rather than to trigger an interrupt. Then you will get a reset to the standard initialisation location, not the bootloader address.