feilipu/Arduino_FreeRTOS_Library #40: FreeRTOSConfig file per project

fjrg76mx, 2018-03-30:
There should be a way to set such a file on a per project basis, instead of per platform. It's not easy even when compiling Arduino on the command line, but the newbies (I mean, the majority of users) might not find their way.

gpb01, 2018-04-19:
Totally agree, like in other platforms, FreeRTOSConfig.h should be per project and not global for all applications.  Normally, for each project, you need/want to customize the file to include/exclude some features or to adjust some values.

feilipu, 2018-06-08:
Yes, this would be good.
But unfortunately the Arduino library capability doesn't support this feature.

Configuration files per sketch for libraries would be a big help.
Please ask the Arduino developers to do it.

closed 2018-06-08 by feilipu