feilipu/Arduino_FreeRTOS_Library #42: Using LinkedList within FreeRTOS results in unsaved values

kfriede, 2018-05-17:
I need to utilize a LinkedList, and rather than do it myself, am using the [LinkedList arduino library here.](https://github.com/ivanseidel/LinkedList).  

The library works in a standard Arduino (loop) configuration, but when I put it into the RTOS, it fails to save values.  The size() counter increments as if it is saving, but all values returned are '0'.  

I initially expected this was an implementation issue with the library, but trying out [Qlist](https://github.com/SloCompTech/QList) yields the same results.  

Is there something special I need to do for a LInkedList to properly handle saving values in FreeRTOS?  

FreeRTOS Code:
```
#include <Arduino_FreeRTOS.h>
#include <LinkedList.h>

void TaskTest( void *pvParameters );

void setup() {

  Serial.begin(115200);
  
  while (!Serial) {
    ;
  }

  xTaskCreate(
    TaskTest
    ,  (const portCHAR *)"Test"
    ,  128
    ,  NULL
    ,  2
    ,  NULL );
}

void loop()
{
  
}

void TaskTest( void *pvParameters __attribute__((unused)) )
{
  LinkedList<byte> list;

  for (int i = 0; i < 256; i++) {
    list.add(i);
    Serial.print(i);
    Serial.print(',');
    Serial.print(list.get(list.size() - 1));
    Serial.print(',');
    Serial.println(list.size());
  }
  
  for (;;)
  {
    vTaskDelay(1);
  }
}
```

FreeRTOS Code Output (truncated):
```
248,0,249
249,0,250
250,0,251
251,0,252
252,0,253
253,0,254
254,0,255
255,0,256
```

Standard Arduino Code:
```
#include <LinkedList.h>

void setup() {
  Serial.begin(115200);
  
  while (!Serial) {
    ;
  }

}

void loop() {
  LinkedList<byte> list;

  for (int i = 0; i < 256; i++) {
    list.add(i);
    Serial.print(i);
    Serial.print(',');
    Serial.print(list.get(list.size() - 1));
    Serial.print(',');
    Serial.println(list.size());
  }
  
  for (;;)
  {
  }

}
```

Standard Arduino Code Output (truncated):
```
249,249,250
250,250,251
251,251,252
252,252,253
253,253,254
254,254,255
255,255,256
```

kfriede, 2018-05-17:
This occurs on both 328P and 2560

FrozenDroid, 2018-05-29:
Have you tried using critical sections? These libraries might not support reentrancy

feilipu, 2018-06-08:
I'm not sure of the way that C++ caches values, but I'm strongly suspicious that local variables from one task (and therefore one stack), won't be automatically copied to the similarly named local variable on another stack in a different task.

If you find it works within one task, either within the loojp or another task, but then fails if you try to do it across different tasks, then that is your problem solved.

I suggest putting the linked list declaration as a global, then the variables will be defined in the data section, and should be common across all tasks.

closed 2018-07-20 by feilipu