feilipu/Arduino_FreeRTOS_Library #45: Help needed, Cannot make FreeRTOS and Udp work together

apurv1012, 2018-07-07:
Hello, Thanks for sharing your work
I am having some difficulty,
I am working on WiFi module(Adafruit WINC 1500) where i have used WIFI101 library to communicate through Udp. I can receive and send data using WIFIUdp without freertos library that is by keeping Udp.parsepacket() in loop. But as soon as i include "#include <Arduino_FreeRTOS.h>" i stop receiving Udp packets. 
Please suggest.

feilipu, 2018-07-15:
Sorry, but haven't used used this Adafruit WINC 1500 shield so I can't comment directly.

I do note [this comment](https://learn.adafruit.com/adafruit-winc1500-wifi-shield-for-arduino/overview) on the Adafruit web site *"Right now the Atmel-supplied library works best with SAMD21-based boards like the Arduino Zero or Metro M0 Express, or the Arduino Mega. It works OK on Uno but uses nearly all the memory so it may be a challenge to build complex projects."*

I suggest that the issue you're facing is memory limitations, particularly if you're using an Uno as the platform. FreeRTOS segregates local variable memory on the stack based on the provision specified at task creation. If you're not allowing enough stack for the task running `Udp-parsepacket()`, then it will fail.

In projects I've worked on in the past I have found that the Uno is only satisfactory for "playing", and not for use with major projects. For example, I couldn't use the SD library at the same time that I used the W5100 library, for example. It as because of these limitations (together with limitations in shield compatibility in the Mega) that lead me to build the [Goldilocks Analogue](https://www.tindie.com/products/feilipu/goldilocks-analogue/), with 16kB of RAM (up from 2kB in the Uno, and 8kB on the Mega).

feilipu, 2018-07-16:
It sounds like you've already lots of experience with the tools.
As I noted, I haven't used either of those libraries so it is impossible for me to be specific with help.

But, in the past my experience is that issues with the FreeRTOS scheduler running are always:
1. Something is blocking the scheduler from running. Somewhere there is a busy-wait `while(1)` or `for(;;)` in the library.
2. Memory issues. A stack is overwriting one or more critical memory areas. Usually with this library is is using the idle task (or `loop()` ) to do something.

Otherwise, I suggest trying to add one small piece or other until the system fails. Then you will know what caused the issue, or at least have an indicator as to where the problems lie.

closed 2018-07-20 by feilipu