feilipu/Arduino_FreeRTOS_Library #52: pcTaskGetName don't work.

nopnop2002, 2019-03-21:
pcTaskGetName don't work.

Environment:
Arduino-IDE 1.8.5 + ATMEGA2560

code:

	#include <Arduino_FreeRTOS.h> // https://github.com/feilipu/Arduino_FreeRTOS_Library

	void setup() {
	  Serial.begin(115200);
	  while(!Serial);
	  Serial.println("Serial Start");
	}

	void loop() {
	  static int counter=0;
	  TickType_t nowTick;
	  UBaseType_t Priority;
	  
	  Serial.print("counter="); 
	  Serial.println(counter);

	  nowTick = xTaskGetTickCount();
	  Serial.print("nowTick=");
	  Serial.println(nowTick);

	  Priority = uxTaskPriorityGet(NULL);
	  Serial.print("Priority=");
	  Serial.println(Priority);

	  Serial.print("TaskName=");
	  Serial.println(pcTaskGetName(NULL));

	  counter++;
	  delay(1000);
	}

result:

	nowTick=2192
	Priority=0
	TaskName=IDLE
	counter=39
	nowTick=2250
	Priority=0
	TaskName=IDLE
	counter=40
	nowTick=2308
	Priority=0
	TaskName=IDLE
	counter=41
	nowTick=2366
	Priority=0
	TaskName=IDLE

feilipu, 2019-03-26:
It is likely that both these issues are caused by a stack overflow.

The `printf()`, and more particularly `Serial.print()`, functions require large amounts of stack.
The initialisation of `Serial` requires 64 Bytes for each of the [Tx](https://github.com/arduino/ArduinoCore-avr/blob/b7c607663fecc232e598f2c0acf419ceb0b7078c/cores/arduino/HardwareSerial.h#L46) and [Rx](https://github.com/arduino/ArduinoCore-avr/blob/b7c607663fecc232e598f2c0acf419ceb0b7078c/cores/arduino/HardwareSerial.h#L53) buffers. Calling `Serial.print()` in each Task may allocate buffers on the stack each time it is initialised.

Given you're using a ATmega2560, try using 512 bytes of stack for each Task, for example.

If that doesn't work, add a comment here.

nopnop2002, 2019-03-28:
OK
I'll try this weekend.

nopnop2002, 2019-03-31:
I give 2048 byte of stack for each task.
But it don't work.

	#include <Arduino_FreeRTOS.h>

	// Task Body
	void task(void *pvParameters)
	{
	#if 0
	  Serial.print("xTaskGetTickCount=");
	  Serial.println(xTaskGetTickCount());
	#endif

	  printf("pcTaskGetName=[%s]\n",pcTaskGetName(NULL));
	  Serial.print("pcTaskGetName=");
	  Serial.println(pcTaskGetName(NULL));
	  pinMode(LED_BUILTIN, OUTPUT);
	  for (;;) // A Task shall never return or exit.
	  {
	    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
	    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
	    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
	    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
	  }
	  vTaskDelete( NULL );
	}

	//------------------------------------------------------------------------------
	void setup() {
	  portBASE_TYPE xTask1, xTask2;

	  Serial.begin(115200);
	  Serial.println("setup() start");
	  Serial.print("configTICK_RATE_HZ:");
	  Serial.println(configTICK_RATE_HZ);
	  Serial.print("portTICK_PERIOD_MS:");
	  Serial.println(portTICK_PERIOD_MS);
	  Serial.print("freeRTOS version:");
	  Serial.println(tskKERNEL_VERSION_NUMBER);

	  xTask1 = xTaskCreate(task, "Task1", 2048, NULL, 2, NULL);
	//  xTask2 = xTaskCreate(task, "Task2", configMINIMAL_STACK_SIZE, NULL, 2, NULL);

	  /* Check everything was created. */
	  configASSERT( xTask1 );
	//  configASSERT( xTask2 );

	// Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
	}

	//------------------------------------------------------------------------------
	// WARNING loop() called from vApplicationIdleHook(), so don't use this function.
	// loop must never block
	void loop() {
	  // Not used.
	}

feilipu, 2020-02-17:
Tested working with below code.

The `printf()` statement was stopping things working.
I didn't bother totest, but usually it needs the `#include <stdio.h>` to work.

```C++

#include <Arduino_FreeRTOS.h>

static void task( void *pvParameters );

// Task Body
void task(void *pvParameters)
{
  Serial.print("pcTaskGetName=");
  Serial.println(pcTaskGetName(NULL));

  pinMode(LED_BUILTIN, OUTPUT);
  for (;;) // A Task shall never return or exit.
  {
    Serial.print(F("xTaskGetTickCount="));
    Serial.println(xTaskGetTickCount());
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
  }
  vTaskDelete( NULL );
}

//------------------------------------------------------------------------------
void setup() {
  portBASE_TYPE xTask1, xTask2;

  Serial.begin(38400);
  Serial.println(F("setup() start"));

  Serial.print(F("configTICK_RATE_HZ:"));
  Serial.println(configTICK_RATE_HZ);
  Serial.print(F("portTICK_PERIOD_MS:"));
  Serial.println(portTICK_PERIOD_MS);
  Serial.print(F("freeRTOS version:"));
  Serial.println(tskKERNEL_VERSION_NUMBER);

  xTask1 = xTaskCreate(task, "Task1", 256, NULL, 2, NULL);
//  xTask2 = xTaskCreate(task, "Task2", 256, NULL, 2, NULL);

  /* Check everything was created. */
 configASSERT( xTask1 );
//  configASSERT( xTask2 );

// Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
}

//------------------------------------------------------------------------------
// WARNING loop() called from vApplicationIdleHook(), so don't use this function.
// loop must never block
void loop() {
  // Not used.
}

```

nopnop2002, 2020-02-18:
printf crashes the system.
And vTaskDelay () also crashes the system.

```
#include <Arduino_FreeRTOS.h>
#include <task.h>

void ConsumptionTick(int delay) {
    TickType_t startTick;
    TickType_t endTick;
    TickType_t nowTick;
    startTick = xTaskGetTickCount();
    endTick = startTick + delay;
    //printf("startTick=%d endTick=%d\n",startTick,endTick);
    while(1) {
      nowTick = xTaskGetTickCount();
      if (nowTick > endTick) break;
    }
}

// Task Body
void task(void *pvParameters)
{
  UBaseType_t prio;
  TickType_t nowTick;

  Serial.print("start pcTaskGetName=");
  Serial.print(pcTaskGetName(NULL));
  Serial.print(" uxTaskPriorityGet=");
  Serial.print(uxTaskPriorityGet(NULL));
  Serial.print(" xTaskGetTickCount=");
  Serial.println(xTaskGetTickCount());

  ConsumptionTick(200);

  Serial.print("end pcTaskGetName=");
  Serial.print(pcTaskGetName(NULL));
  Serial.print(" xTaskGetTickCount=");
  Serial.println(xTaskGetTickCount());

  vTaskDelete( NULL );
}

//------------------------------------------------------------------------------
void setup() {
  TaskHandle_t xHandle1 = NULL;
  TaskHandle_t xHandle2 = NULL;
  BaseType_t xReturned;
  
  Serial.begin(115200);
  Serial.println("setup() start");
  Serial.print("configTICK_RATE_HZ:");
  Serial.println(configTICK_RATE_HZ);
  Serial.print("portTICK_PERIOD_MS:");
  Serial.println(portTICK_PERIOD_MS);
  Serial.print("configMAX_PRIORITIES:");
  Serial.println(configMAX_PRIORITIES);
  Serial.print("freeRTOS version:");
  Serial.println(tskKERNEL_VERSION_NUMBER);

  xReturned = xTaskCreate(task, "myTask1", 256, NULL, 2, &xHandle1);
  if( xReturned != pdPASS ) {
    Serial.println("myTask1 xTaskCreate fail");
  }

  // delay 16mSec
  //vTaskDelay(16 / portTICK_PERIOD_MS);
  vTaskDelay(10);

  xReturned = xTaskCreate(task, "myTask2", 256, NULL, 3, &xHandle2);
  if( xReturned != pdPASS ) {
    Serial.println("myTask2 xTaskCreate fail");
  }

// Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
}

//------------------------------------------------------------------------------
// WARNING loop() called from vApplicationIdleHook(), so don't use this function.
// loop must never block
void loop() {
  // Not used.
}
```

feilipu, 2020-02-19:
I mentioned above that [to use `printf()` you need to include the `stdio.h`](https://www.tutorialspoint.com/c_standard_library/c_function_printf.htm) header in your code, or it won't work.

Also, `vTaskDelay()` is not "crashing the system". I demonstrated it working in the code above.

Please, read the instructions on how to use the library from the FreeRTOS web site, and use working  examples as a tutorial.

nopnop2002, 2020-02-19:
In my environment, even if stdio.h is added, printf () crashes.
And vTaskDelay () also crashes.
Above code works fine with esp-idf.

My environment:
Arduino-IDE 1.8.12 + ATMega2560
```
name=FreeRTOS
version=10.2.0-2
author=Richard Barry <info@freertos.org>
maintainer=Phillip Stevens <phillip.stevens@gmail.com>
sentence=Real Time Operating System implemented for AVR (Uno, Leonardo, Mega).
paragraph=The primary design goals are: Easy to use, Small footprint, Robust. Uses Watchdog Timer for 15ms resolution. Slow blink = stack overflow. Fast blink = heap malloc() failure.
category=Timing
url=https://github.com/feilipu/Arduino_FreeRTOS_Library
architectures=avr
licence=MIT
includes=Arduino_FreeRTOS.h
```

feilipu, 2020-02-19:
I'd note that a quick Google search would inform that `printf()`doesn't work at all in the Arduino IDE.
Apologies that I didn't do that for you earlier.

Open the `File/Examples/FreeRTOS/Blink_AnalogRead` example. Compile and upload it. Note the LED is flashing. Note that analogue samples are printed on the serial port. Note that `vTaskDelay()` doesn't crash. Therefore, note that there's something you need to fix with your sketch.

feilipu, 2020-02-19:
You can't do a delay depending on the scheduler, before the scheduler has been started...
```
  xTask1 = xTaskCreate(task, "Task1", 256, NULL, 2, NULL);
  vTaskDelay(10);
  xTask2 = xTaskCreate(task, "Task2", 256, NULL, 2, NULL);
```
Read the comment at the bottom of the set up...
```
// Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
```

nopnop2002, 2020-02-19:
OK!!

I understood everything.
Thank you for your support.

closed 2020-02-19 by nopnop2002