feilipu/Arduino_FreeRTOS_Library #55: Redefine something in the "FreeRTOSConfig.h" on a user sketch

JoshuaKim6502, 2019-08-27:
Hi Phillip,
I enjoy using the FreeRTOS on the Arduino Uno.  I understand why the header file "FreeRTOSConfig.h" is not in an individual project but the Ardunio/libraries/FreeRTOS/src.

What if I put a needed line to my source sketch, while the FreeRTOSConfig.h is intact? I think it works. Correct?

For example, I copy this line to my sketch;

>#define configMAX_PRIORITIES    (1+5)

while the FreeRTOSConfig.h has the original;

>#define configMAX_PRIORITIES    3

It looks work when I compile as follows:

>:\Users\Joshua\iWork\Ardunio\sketch_aug08a\sketch_aug08a.ino:6:0: warning: >"configMAX_PRIORITIES" redefined
>
 >#define configMAX_PRIORITIES (1+5)
 >^
>In file included from >C:\Users\Joshua\Documents\Arduino\libraries\FreeRTOS\src/Arduino_FreeRTOS.h:56:0,
>                 from C:\Users\Joshua\iWork\Ardunio\sketch_aug08a\sketch_aug08a.ino:2:
>C:\Users\Joshua\Documents\Arduino\libraries\FreeRTOS\src/FreeRTOSConfig.h:51:0: note: this >is the location of the previous definition
>
> #define configMAX_PRIORITIES                3
> ^

feilipu, 2019-08-28:
Joshua,
I'm not sure what will happen.
I suggest that you assign a variable in your sketch with `configMAX_PRIORITIES`, and then print it.
First do it without redefining in your sketch, to ensure you're seeing the original number `(1+5)`, and then again with your redefinition to ensure you see the new number `3`.
Best regards, Phillip

JoshuaKim6502, 2019-08-28:
Thanks for the comment.  I tested again as you recommend.
Without redefining the `configMAX_PRIORITIES` value, my code prints the original **`3`** as expected.
Once I redefine the value to `6` in my source, the result printed is **`6 = (1+5)`**.
So the redefinition on the user source looks work.

Thanks again, Joshua

JoshuaKim6502, 2019-08-30:
I misunderstood it. After precise diagnostic concerning the `configMAX_PRIORITIES` value, I realized that redefining it on the user source is meaningless even though it looks to be changed.
The original `<FreeRTOSConfig.h>` dominates compiling kernel codes first off, while any redefined on the user source appears last.

closed 2019-09-02 by JoshuaKim6502