feilipu/Arduino_FreeRTOS_Library #59: Enhance Documentation

SpamWall, 2019-09-19:
The documentation is a little unclear about how to upgrade to the latest FreeRTOS version.  The links take us to the FreeRTOS page, but what then?  Do we just overlay the FreeRTOS files over the Arduino_FreeRTOS src and rebuild?  Does Arduino_FreeRTOS test new versions of FreeRTOS?

Having a couple of paragraphs on this would be very helpful.

JoshuaKim6502, 2019-09-20:
Please open the Arduino IDE and select  Tools--> Manage Libraries; search "FreeRTOS" and install Richard Barry's.  
[image]

JoshuaKim6502, 2019-09-20:
This page would help you to start;
https://feilipu.me/2015/11/24/arduino_freertos/

feilipu, 2019-09-21:
I always try to stay up with the latest version of FreeRTOS, so all that is needed is to install the library using the Aduino Library Manager. Then you're good to go.

As @JoshuaKim6502 has pointed out, there's further info on the web site.

closed 2019-09-21 by feilipu