feilipu/Arduino_FreeRTOS_Library #61: pdMS_TO_TICKS() overflows

manuelferreira65, 2019-11-12:
#define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( ( ( TickType_t ) ( xTimeInMs ) * ( TickType_t ) configTICK_RATE_HZ ) / ( TickType_t ) 1000 ) )

If TickType_t defined as uint16_t  xTimeInMs * configTICK_RATE_HZ overflows with xTimeInMs = 1058.

feilipu, 2020-01-05:
I don't think it was expected that `TickType_t` would be a 16 bit type, hence top line, before divide, overflows.

Try removing the casts from the top line. Richard may already have fixed this. I'll check.

feilipu, 2020-02-17:
Resolved with 10.3.0-1, [here](https://github.com/feilipu/Arduino_FreeRTOS_Library/commit/a28b4794247840348045443be487c9ed6a26e4b4#r37324607).

closed 2020-02-17 by feilipu

feilipu, 2020-02-26:
Resolved finally [with this code](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/projdefs.h#L41).

``` c
#if configUSE_16_BIT_TICKS == 1
    #define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( ( ( uint32_t ) ( xTimeInMs ) * ( TickType_t ) configTICK_RATE_HZ ) / ( TickType_t ) 1000 ) )
#else
    #define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( ( ( TickType_t ) ( xTimeInMs ) * ( TickType_t ) configTICK_RATE_HZ ) / ( TickType_t ) 1000 ) )
#endif
```