feilipu/Arduino_FreeRTOS_Library #63: malloc() not working inside tasks

leadpogrommer, 2020-03-07:
Hello.
If you try to use malloc() inside task, it will always return 0.

This simple code
`#include <Arduino_FreeRTOS.h>

void *bugTask(void* params){
  void *ptr = pvPortMalloc(1);
  while(true);
}

void setup() {
  xTaskCreate(bugTask, "bug", 150, NULL, 1, NULL);
}

void loop() {}`
causes LED to flash rapidly => heap overflow

Tested on Arduino UNO and Mega

I've found this article [https://mcuoneclipse.com/2014/03/16/freertos-malloc-and-sp-check-with-gnu-tools/](url), but i dont'know how to fix it in described way

feilipu, 2020-04-03:
Sorry about the delay. This issue confirmed. I'll look into fixing it.
Thanks for reporting.

feilipu, 2020-04-03:
Ok, so the issue is caused because the `__malloc_heap_end` is defined to be `0`, [here](http://svn.savannah.gnu.org/viewvc/avr-libc/trunk/avr-libc/crt1/gcrt1.S?view=markup) on line 215.

The bug is caused [here](http://svn.savannah.gnu.org/viewvc/avr-libc/trunk/avr-libc/libc/stdlib/malloc.c?view=markup) on line 166, where the `__brkval` (which has been set from prior Task TCB allocation) will be greater than the `SP` of the running Task.

So either, the `__malloc_heap_end` needs to be redefined sensibly during linking. I'm not sure how to do this in the most robust and flexible manner across a number of different machine types.

Possibly, assuming a small startup stack would be a good way?
This would set it to be `RAMEND - 128` for example.

Or, just write another `malloc()` and use the linker override as per the suggestion.
The relevant section is...

```c
/*
 * Step 3: If the request could not be satisfied from a
 * freelist entry, just prepare a new chunk.  This means we
 * need to obtain more memory first.  The largest address just
 * not allocated so far is remembered in the brkval variable.
 * Under Unix, the "break value" was the end of the data
 * segment as dynamically requested from the operating system.
 */
if (__brkval == 0)
    __brkval = __malloc_heap_start;
avail = RAMEND - __malloc_margin - __brkval;
```

feilipu, 2020-04-03:
Fixed with [Release 10.3.0-4](https://github.com/feilipu/Arduino_FreeRTOS_Library/releases/tag/10.3.0-4).
Allowing so much initial stack is wasteful, but this is the simplest fix.
Please confirm it works as expected for you, and close the issue when ready.

leadpogrommer, 2020-04-04:
I'll check it in couple of days

leadpogrommer, 2020-04-05:
[Release 10.3.0-4](https://github.com/feilipu/Arduino_FreeRTOS_Library/releases/tag/10.3.0-4) leads to error 404

leadpogrommer, 2020-04-05:
It works.
Thank you.

closed 2020-04-05 by leadpogrommer

itopaloglu83, 2021-12-01:
Hello @feilipu, I'm just curious and I have a question about this bug. I hope you can find the time to entertain it.

My understanding is that when `__malloc_heap_end` is set to `0` it serves as a signal for `malloc` to dynamically check for the end of the heap by subtracting a safety margin (`__malloc_margin`) from the stack pointer. However, inside a task, the stack pointer is pointing to a location inside the heap itself and hence causing the malloc to assume that memory is exhausted. By manually defining `__malloc_heap_end` we are telling effectively reserving a stack of `__malloc_margin` bytes and telling the `malloc` where the heap ends.

Would you say that this is an accurate picture of the issue?

feilipu, 2021-12-01:
> My understanding is that when `__malloc_heap_end` is set to `0` it serves as a signal for `malloc` to dynamically check for the end of the heap by subtracting a safety margin (`__malloc_margin`) from the stack pointer. However, inside a task, the stack pointer is pointing to a location inside the heap itself and hence causing the malloc to assume that memory is exhausted. By manually defining `__malloc_heap_end` we are effectively reserving a stack of `__malloc_margin` bytes and telling the `malloc` where the heap ends.
> 
> Would you say that this is an accurate picture of the issue?

Yes.

Let me expand. Each task is provided with a working memory and its own stack on initialisation from the heap, and this will be lower in memory than the normal C stack (at the top of memory). So when a Task is running the SP will be set quite low, and certainly below the end of the current C heap (from which Task working memory and stack is allocated).

So in order to get `malloc()` to behave correctly (and allocate from true free memory, not Task assigned memory) we need to tell it where the true end of heap is located via `__brkval`.

Probably less well explained than your definition, but at least how I understand it to work. Hope that's useful.

itopaloglu83, 2021-12-01:
> So in order to get `malloc()` to behave correctly (and allocate from true free memory, not Task assigned memory) we need to tell it where the true end of heap is located via `__brkval`.

This is the part that made me understand not only the issue but what purpose the break value (`__brkval`) actually serves. It's quite a technical issue. Thank you for the fix and of course the library!