feilipu/Arduino_FreeRTOS_Library #66: Storage size problem

boralbgra, 2020-03-31:
**Describe the issue**
The Sketch uses 47,228 bytes (146%) of the program storage space. The maximum is 32256 bytes.
text section exceeds available space in board
Global variables use 1574 bytes (76%) of dynamic memory, leaving 474 bytes for local variables. The maximum is 2048 bytes.
Very cool program: visit http://www.arduino.cc/en/Guide/Troubleshooting#size to see how to reduce it.
Compiling error for the Arduino Uno board.

**Expected behavior**
I would like to be able to compile programs cause I cant compile anything.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10]

**Additional context**
Add any other context about the problem here.

feilipu, 2020-04-03:
I have tried to replicate the issue (see screenshot below) using the most recent IDE version (but that shouldn't matter), and I can't.

Perhaps you can try installing the IDE again? But, I can't advise that will actually help.

[image]

feilipu, 2020-04-05:
Please reopen if issue persists.

closed 2020-04-05 by feilipu