feilipu/Arduino_FreeRTOS_Library #76: Just including Wire.h library make the sheduler stop working.

JBRO8081, 2020-06-28:
Code 1
/*---------------------------------#includes---------------------------------*/
#include <SPI.h>
#include <Wire.h>

#include <Arduino_FreeRTOS.h>
#include <queue.h>>

/*--------------------------------#defines-----------------------------------*/

#define PIN_TEMP_SENSOR A0
#define PIN_TEST_ANA    A3
#define PIN_RELAY_GND   5
#define PIN_RELAY_EXT   6
#define PIN_SPI_CS_DAC  9
#define PIN_SPI_CS_ADC  10       
#define ADC_VREF    2500 // 2500mV
#define ADDR_DIGITAL_POT 0x29
#define I2C_CLK_RATE 400000
#define SERIAL_BAUD_RATE 9600

/*
 * Serial Protocol
 * ,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
 * |Sensor Name|Channel No|Val LSB|Val MSB|\n|
 * '''''''''''''''''''''''''''''''''''''''''''
 * @Sensor Name : T for temp, V for ADC_V ,C for ADC_C,P for Pot val,
 *                D for DAC Val.
 * @Channel No  : Channel number of sensor if applicable or default it to zero.
 * @Val LSB     : Lower byte of raw value from sensor.
 * @Val MSB     : Upper byte of raw value from sensor.
 * @\n          : Termination byte to be new line.
 * Please Note that Raw value proessing is expected to taken care by host due 
 * to Limited processing capability of atmega 328P. Total byte size is limited 
 * to 5 byte for performance reasons.
 */
#define QUEUE_SERIAL_SEND_DEPTH 10
#define QUEUE_SERIAL_SEND_BYTE_MAX_SIZE 5

#define TASK_SERIAL_SEND_STACK 128
#define TASK_SERIAL_SEND_PRIORITY 2

#define TASK_SERIAL_RECV_STACK 128
#define TASK_SERIAL_RECV_PRIORITY 1

#define TASK_VIT_READ_STACK 128
#define TASK_VIT_READ_PRIORITY 3

#define TASK_HOUSE_KEEP_STACK 128
#define TASK_HOUSE_KEEP_PRIORITY 0

#define TASK_MODE_SWITCH_STACK 128
#define TASK_MODE_SWITCH_PRIORITY 0

/*--------------------------------Global Variables---------------------------*/

/*
 * This Queue will be used to collect data from various 
 * sensor and send to host. 
 */
QueueHandle_t SerialDataQueue;

/*----------------------------------Functions--------------------------------*/

void setup() 
{
  
/*
 * Settings for different IO pins according to its uses and their defualt
 * states.
[... 242 lines cut ...]

  for (;;)
  {
    if (xQueueReceive(SerialDataQueue, buf, portMAX_DELAY) == pdPASS)
    {
      Serial.write(buf, QUEUE_SERIAL_SEND_BYTE_MAX_SIZE);
    }
  }
}

void TaskDataSend(void * pvParameters)
{
  uint8_t buf[] = "HAI00";

  while (!Serial)
  {
    vTaskDelay(1);
  }

  for (;;)
  {
    xQueueSend(SerialDataQueue, buf, portMAX_DELAY);
  }
}
I was trying to interface atmega328 with MCP4651T Digital pot working with I2C(code 1). I find weird that including Wire.h make my scheduler stopped working. The device is working fine without any problem with Code 2. I don't know what is causing this issue. Could you please assist me.

feilipu, 2020-06-28:
I don't have time today to look at this in detail and try to replicate, but my first guess is that you've run out of RAM, and somewhere a stack is being overwritten.

The ATmega328 doesn't have much to go around. If you have a ATmega2650, then try that to see if the problem doesn't show. 

Also try being more conservative with some of the stack sizes for your null tasks (HouseKeep, SerialReceive, ModeSwitch) currently, say around 96 bytes. See if that changes anything.

JBRO8081, 2020-06-29:
As you said I was running out of the stack. I didn't mount an LED on pin 13 to get a visual indication.Sorry my bad.

closed 2020-06-29 by JBRO8081