feilipu/Arduino_FreeRTOS_Library #78: vTaskDelay not working

RogerRU, 2020-07-21:
Atmega1281
Run standart example Blink_AnalogRead, all tasks work fine, but vTaskDelay not working with any arguments
Tried changing portUSE_WDTO - no changes :-(

feilipu, 2020-07-22:
What do you mean, "not working with any arguments".
Is it not compiling? Or, is it not delaying by the correct amount?
Or, is the delay the same no matter what argument you use?

It is possibly something to do with the ATmega1281 MCU and setting the FCPU value correctly.
What is your platform? Where is the ATmega1281 mounted? Is it from any company?

RogerRU, 2020-07-22:
It looks like I understood what was the matter, and it’s not about FreeRTOS. The controller cyclically reboots every 10-20 milliseconds.
Initially, my board was made for ATmega128 and worked good, but then I decided to replace it with Atmega1281 to try FreeRTOS in my project. As far as I understand, the controllers are fully compatible in pins and wiring diagrams, are there any slight differences in registers.
My whole board is 3.3v and I thought that the problem was that the Atmega1281 does not work well at 16MHz at 3.3v, I replaced the crystal with 8MHz, changed fuses, but the problem remained. 
I understand that this is offtopic, but can you give advice on how to solve the problem.
Thank.

feilipu, 2020-07-22:
Possibly unrelated, but the ATmega1284p doesn't work well unless full swing oscillator is set in the fuses. Perhaps that's the issue?

And, you've understood all the issues covered in the [migration reference document](http://ww1.microchip.com/downloads/en/AppNotes/doc2581.pdf)?

closed 2020-09-12 by feilipu