feilipu/Arduino_FreeRTOS_Library #93: Teensy support

gpb01, 2021-01-08:
Hi Phillip,

standard AVR board are quite limited and FreeRTOS™ certainly cannot give the best of itself on a few Kbytes SRAM board with an 8 bit MCU running at 16 MHz. :-)

I would like to ask you if will be possible to add support fo the Teensy boards using **Teensyduino** (_Arduino IDE with Teensy core_) ... with the last Teensy 4.1 we have a very powerful board running at **600MHz**, with a lot or SRAM and flash memory !

If can help, here : https://github.com/tsandmann/freertos-teensy you can find already a porting for teensy boards; maybe looking at "porting files" can be useful for you to implement.

Thanks a lot in advance,

Sincerely,
Guglielmo

feilipu, 2021-01-09:
Hi Guglielmo,
thanks for your input and suggestions.

> Standard AVR board are quite limited and FreeRTOS™ certainly cannot give the best of itself on a few Kbytes SRAM board with an 8 bit MCU running at 16 MHz.

But, I beg to differ. FreeRTOS on ATmega and other 8-bit (e.g. [Z80](https://github.com/feilipu/z88dk-libraries/tree/master/freertos)) platforms is very capable and works with no issues. In fact, IMHO, one of the strengths of FreeRTOS is its ability to be successfully used on very resource limited platforms.

The purpose of this repository is to offer the lowest common denominator entry method for new users to experience. The Arduino Team have used the UNO (and predecessors) as their entry level platform for such a long time now, and this entry level is designed for these users. Once people get experience, they will of course move on to more capable platforms, and more complex systems.

> I would like to ask you if will be possible to add support fo the Teensy boards using Teensyduino (Arduino IDE with Teensy core)...
> ...here : https://github.com/tsandmann/freertos-teensy you can find already a porting for teensy boards.

I see that the work to support FreeRTOS on Teensy boards has already been done, and it would not be sensible to reproduce @tsandmann work here. Is there some benefit to copying this repos work here?

gpb01, 2021-01-09:
Hi Phillip,
thanks for your reply.

> I see that the work to support FreeRTOS on Teensy boards has already been done, and it would not be sensible to reproduce @tsandmann work here. Is there some benefit to copying this repos work here?

@tsandmann work is for **PlatformIO** but I don't find how to use in Teensyduino (_Arduino IDE + Teensy cores_). Your library is already for **Arduino IDE** ...  if you add the "_porting_" files for Teensy, will be possible to use your library on Teensyduino :-)

tsandmann, 2021-01-09:
Hi Guglielmo,

basically my library should also be usable with the Arduino IDE. However, it needs some (minor) changes of the Teensy cores library, mainly because the systick and software interrupts that the ARM FreeRTOS port needs are also used internally by the Teensy cores library. The PlatformIO examples have a dependency on a patched cores library that handles these conflicts.

Therefore just adding the *porting* files to an Arduino library isn't sufficient. If you want to use FreeRTOS with Teensyduino, you would have to patch the cores library provided by Teensyduino yourself. Maybe the patches will be integrated upstream in the future, but so far I haven't tested them sufficiently for the non-FreeRTOS case.

Regards,
Timo

gpb01, 2021-01-09:
Thanks Timo, a this point, I think is better that I open an issue on your github and Pilliph can close this one in a library more AVR oriented  ;-)

closed 2021-01-10 by feilipu