feilipu/Arduino_FreeRTOS_Library #97: A Few Doubts

Coder-X15, 2021-01-25:
I have a few doubts regarding this project, which I'd like to get cleared:
1. Does the ``xTaskCreate()`` function support functions that are not of the ``void`` type ? 
2. Is it capable of speeding up custom-made GUI ?
3. Where can I get a better reference guide on this library and its functions ?

Thanks in advance ! :)

feilipu, 2021-01-25:
1. Does the `xTaskCreate()` function support functions that are not of the void type? - The reference for is [here](https://www.freertos.org/a00125.html) in the FreeRTOS documentation. Everything about FreeRTOS is there to read.
2. Is it capable of speeding up custom-made GUI? - Programming a GUI is unrelated to FreeRTOS.
3. Where can I get a better reference guide on this library and its functions? - The extensive documentation for FreeRTOS is available [here](https://www.freertos.org/FreeRTOS-quick-start-guide.html). Questions arising can be asked at the [FreeRTOS community forum](https://forums.freertos.org/). It is best to keep this Issues register purely for library implementation issues (bugs or enhancements). Or use the Discussions Tab above for discussion.

closed 2021-01-25 by feilipu

Coder-X15, 2021-01-25:
Thanks... seems like my second doubt was stupid, but why I actually asked it was whether it could be used to run two different widgets on a screen at the same time by using the capabilities of this marvellous project ':)

Coder-X15, 2021-01-25:
Also, thanks for labeling this as a discussion.. I didn't know how to label these 😄

feilipu, 2021-01-25:
> ...whether it could be used to run two different widgets on a screen at the same time?

Typically, one Task might be set up to manage a piece of hardware or logical interface (i.e. the screen). Then you'd use message passing via [Queues](https://www.freertos.org/Embedded-RTOS-Queues.html) to inform that task that the widgets need to be updated, using one or more Tasks to generate the messages if you have separate sources.

One Task might read the temperature, and another might handle the keyboard. Inputs from these two Tasks could change screen widgets via queue based message passing.

But, that is a very general answer to a very general question.

Coder-X15, 2021-01-25:
Ok thanks....