feilipu/Arduino_FreeRTOS_Library #100: STM32FreeRTOS myservo can not work.

Nanmei, 2021-03-11:
[image]
[image]
`#include <STM32FreeRTOS.h>
#include <Servo.h>    
Servo myservo; 
int pos = 0; 

void vTask1(void *pvParameters){
 myservo.attach(PA7);
  for(;;)
    {
   for(pos = 0; pos < 90; pos += 1){           
      myservo.write(pos);         
       vTaskDelay(20 / portTICK_PERIOD_MS);                  
   }
     vTaskDelay(255 / portTICK_PERIOD_MS); 
    for(pos = 90; pos>=1; pos-=1) {                                 
       myservo.write(pos);            
       vTaskDelay(20 / portTICK_PERIOD_MS);                 
    } 
    vTaskDelay(255 / portTICK_PERIOD_MS);
      
    }
}
void setup() {
  xTaskCreate(vTask1,NULL,64,NULL,1,NULL);
  }
void loop(){
  ;
}`

**This is my code , this code can not work in the STM32FreeRTOS.But this code can work in the arduino.I can't find the reason.**

feilipu, 2021-03-11:
Sorry, but I can't help you with STM32 code or issues.

Good luck.

closed 2021-03-11 by feilipu

Nanmei, 2021-03-12:
I find the reason thanks