feilipu/Arduino_FreeRTOS_Library #104: what is the difference of three freertos?

youxiaojie, 2021-04-07:
https://github.com/feilipu/Arduino_FreeRTOS_Library
https://github.com/feilipu/miniAVRfreeRTOS
https://github.com/feilipu/avrfreertos
and the original rtos kernel avr ports  https://www.freertos.org/a00098.html ?

feilipu, 2021-04-07:
AVRfreeRTOS contains all the options and library code I use with my Uno, Mega2650, and Goldilocks 1284p platforms. The are options for different memory management and even Xtended RAM support on Mega. It is all I've built up.

MINIfreeRTOS is cut down to the bare minimum, mainly to see which files are needed. It could be used as a base, but is more of a reference than something working.

ArduinofreeRTOS is tuned to work with all ATmega devices (that I can support with avr-libc). But it loses control over many things to fit into Arduino IDE environment. Most critically, the preferred Timer can't be used, as the WDT is all that is available to be system tick. But it works quite well and can be a good start to learn about FreeRTOS.

I initially built ArduinofreeRTOS to enable me to use Arduino libraries and drivers, but not lose my preferred FreeRTOS capabilities.

youxiaojie, 2021-04-07:
take 328p/uno as example,it.can use avrfreertos and arduinofreertos. in arduino ide in arduino libs select arduinofreertos, inavrgcc with avrlib using avrfreertos? right?

and because of arduino libs , the prefered timer can't be used?

feilipu, 2021-04-07:
AVRfreeRTOS cannot be used with Arduino IDE. I use Eclipse IDE, and hence there are no make files etc.

Arduino configures all timers in its standard libraries. If you steal one for FreeRTOS system tick something in the libraries will break, or the compile will fail.

youxiaojie, 2021-04-07:
furthermore I am using lgt382p a compatible mcu, which has more timer is  it has more timers is it possible to overcome this problem by using other timers?

youxiaojie, 2021-04-07:
> AVRfreeRTOS cannot be used with Arduino IDE. I use Eclipse IDE, and hence there are no make files etc.
 arduinofreerts can cooperate with arduino ide?

youxiaojie, 2021-04-07:
http://marketplace.eclipse.org/content/arduino-create-eclipse-plug

http://marketplace.eclipse.org/content/arduino-eclipse-ide-named-sloeber-product

which one can run your avrfreertos？not arduinofreertos from lib manager.

feilipu, 2021-04-07:
Read the posts here. https://feilipu.me/2011/09/22/freertos-and-libraries-for-avr-atmega/

closed 2021-04-11 by feilipu