feilipu/Arduino_FreeRTOS_Library #105: FreeRTOS automatically restarts after the last task has finished on Arduino Mega 2560

flo2k, 2021-05-01:
Here, I have a minimal example of FreeRTOS on an Arduino Mega 2560 with one task and the idle hook activated.

```cpp
//FreeRTOS definitions
TaskHandle_t worker_task_handle = NULL;

//prototypes
void worker_task(void* pvParameters);

void setup() {
  //configure serial port
  Serial.begin(9600);
  Serial.println("\r\n\r\nStart...\r\n");

  //create task
  BaseType_t result = xTaskCreate(
    worker_task,        //address of task function
    "worker_task",      //name of task
    128L,               //task stack size in words: 128*2=256 bytes
    NULL,               //task parameters
    2,                  //task priority
    &worker_task_handle //pointer to task handle
  );

  configASSERT(result == pdPASS);           //assert that the task could be created
  configASSERT(worker_task_handle != NULL); //assert that task handle is valid
}

//idle task function
void loop()
{
  //Serial.println("idle...");
}

void worker_task(void* pvParameters) {
  Serial.println("worker_task started...");
  vTaskDelay(pdMS_TO_TICKS(1000));
}
```

When the `worker_task` runs, it prints "worker_task started...", and then it finishes after 1000ms (after the delay). Unexpectedly, after the `worker_task` has finished, the Arduino Mega 2560 restarts. I also use FreeRTOS on an ST NUCLEO L433RC-P (Cortex M4) in combination with the ST HAL driver, and there, the same example does not force a restart.

If I don't start the task in the `setup()` function, then the idle task still works and no restart is performed.

My expected behaviour is, that after the `worker_task` has finished, the idle task should run forever. I don't want that the Arduino Mega 2560 restarts automatically. 

Is there any possibility to change this behaviour or to investigate what causes the restart?

Best,
flo2k

feilipu, 2021-05-01:
Unexpected outcome.
AFK. Will replicate and respond from Monday.

flo2k, 2021-05-01:
If I can help you diagnosing and testing, don't hesitate to contact me.

gpb01, 2021-05-01:
A FreeRTOS tasks **MUST never ending**.

Tipically you have a for (; ;) { ... your task ... } ... for more info: https://www.freertos.org/implementing-a-FreeRTOS-task.html

If you need, you can suspend the task using FreeRTOS vTaskSuspend() function.

Guglielmo

flo2k, 2021-05-01:
@gpb01  Thank you for that hint. I just tested with:
```cpp
void worker_task(void* pvParameters) {
  Serial.println("worker_task started...");
  vTaskDelay(pdMS_TO_TICKS(1000));

  vTaskDelete(nullptr);
}
```
and now, no restart is performed anymore.

Thx a lot, I wasn't aware of that rule.

gpb01, 2021-05-01:
@flo2k : you're welcome !

Guglielmo

flo2k, 2021-05-01:
@feilipu do we need any further investigations? For me, I have a working solution now. If everything is clear, we could close the issue.

feilipu, 2021-05-01:
Please close if you're ok. The resolution is correct, afaik.

closed 2021-05-01 by flo2k