feilipu/Arduino_FreeRTOS_Library #107: vTaskDelay not functioning as expected

s1913388, 2021-08-10:
Code:
```cpp
#include <Arduino.h>
#include <Arduino_FreeRTOS.h>

void setup()
{
    Serial.begin(9600);
    while (!Serial)
    {
        ;
    }
    Serial.print("Setup and loop task running with priority ");
    Serial.println(uxTaskPriorityGet(NULL));
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    Serial.println("OK");
}

void loop() {}

```
Expected (if comment out the vTaskDelay line):
```shell
Setup and loop task running with priority 32
OK
```
Actual output: Weird output.

feilipu, 2021-08-10:
The scheduler is not running until after the end of the setup() function. Therefore calling for a task delay via vTaskDelay is meaningless.

Also no tasks have been created, so except for the idle task running the contents of loop() function, your code has no tasks running with any priority.

Suggestion, read the documentation. Follow the examples provided, and you will find it will be easier.

s1913388, 2021-08-11:
However it runs perfectly with ESP32 in Arduino IDE,
```cpp
#include <Arduino.h>

void setup()
{
    Serial.begin(9600);
    while (!Serial)
    {
        ;
    }
    Serial.print("Setup and loop task running with priority ");
    Serial.println(uxTaskPriorityGet(NULL));
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    Serial.println("OK");
}

void loop() {}

```
It gives expected output:
```shell
Setup and loop task running with priority 1
OK
```

Actually the code is shamelessly copied from the guide [Digi-Key](https://github.com/ShawnHymel/introduction-to-rtos/blob/main/03-task-scheduling-and-management/esp32-freertos-03-demo-prioritization/esp32-freertos-03-demo-prioritization.ino)

@feilipu I am wondering why "OK" is not printed for the Arduino Mega 2560?

feilipu, 2021-08-11:
The ESP32 is a totally different 240MHz / dual core 32-bit thing. Their [IDF uses FreeRTOS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/pthread.html) to schedule WiFI / Bluetooth activities under the cover.
The "OK" is not printed because the `vTaskDelay()` function is waiting for an increment that never happens (because the scheduler is not running). Your results will be completely random at that point.

closed 2021-08-21 by feilipu