feilipu/Arduino_FreeRTOS_Library #108: Is it possible to implement a system timer interrupt version (rather than WDT) compatible with Arduino IDE?

NeoNatural, 2021-09-26:
I find the 15 ms time slice is still too large to meet the requirement of something like robotic control. I noticed another task management library for TaskManagerIO on https://github.com/davetcc/TaskManagerIO, which supports microsecond-level scheduling and works with Arduino IDE. Is there a way to introduce timer 0,1 properly into freertos into Arduino IDE?

feilipu, 2021-09-26:
I have many different options for timers using (Arduino) ATmega boards.
They're all available here at [avrfreertos](https://github.com/feilipu/avrfreertos/tree/master/freeRTOS10xx).

You'll need to be aware that the Arduino IDE will fight you for resources (Timers etc), so you may have to redo configurations after the `setup()` function completes. All the Timers are supported on any of the standard Arduino hardware.

Down to about 1ms is the best timing option, imho.

NeoNatural, 2021-09-26:
Thank you for reply. I have checked the repository I mentioned above. It dose not use any timer at all but depend fully on the arduino loop. I will use the avrfreertos library instead.

closed 2021-09-26 by NeoNatural