feilipu/Arduino_FreeRTOS_Library #110: Generic device support

nicola-lunghi, 2021-11-10:
Hi,
It is possible to make this more generic to make porting to other platforms easier? For example an esp32 support would be difficult?
Thanks,
Nick

gpb01, 2021-11-10:
ESP32 is already based on FreeRTOS™ ... maybe you cabn buy a book on it: https://www.elektor.com/freertos-for-esp32-arduino 😉

feilipu, 2021-11-10:
Yes. That's a good point.

There are quite a few different FreeRTOS options for the ESP32, including using the [Espressif IDK](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html) directly.

Other non ATmega platforms are out of scope for this library though, as there are plenty of options.
This library is really meant only as a very first step in the direction of FreeRTOS and microcontrollers.

closed 2021-11-10 by feilipu

gpb01, 2021-11-10:
Just for completeness and information ... [HERE](https://github.com/tsandmann/freertos-teensy) you can find a port (_also valid for Teensyduino .... Arduino IDE_) for **Teensy 3.5, 3.6, 4.0, 4.1** 🙂

Guglielmo