feilipu/Arduino_FreeRTOS_Library #111: RTOS Example Compile Error

NoSmoke999, 2021-12-06:
**Describe the issue**

C:\Users\LAF\Documents\Arduino\RTOS_example\RTOS_example.ino: In function 'void setup()':
C:\Users\LAF\Documents\Arduino\RTOS_example\RTOS_example.ino:14:15: warning: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
     ,  (const portCHAR *)"Blink"   // A name just for humans
               ^~~~~~~~
RTOS_example:14:9: error: expected primary-expression before 'const'
     ,  (const portCHAR *)"Blink"   // A name just for humans
         ^~~~~

Compiles if that portion of statement is removed. i.e. ",   Blink" only.

**Desktop (please complete the following information):**

 Win 10
Arduino 1.8.16

feilipu, 2021-12-07:
I'm sorry, but I don't recognise the example code.
From the directory structure provided I guess it relates to some other library or example.

NoSmoke999, 2021-12-07:
Probably, I don't recall now where I got the example but it is of no matter as it is easily rectified as noted previously.  I have installed other examples and it works fine.

One question though, are the Arduino FreeRTOS APIs the same as the more general FreeRTOS program?

feilipu, 2021-12-08:
> Are the Arduino FreeRTOS APIs the same as the more general FreeRTOS program?

Yes. Entirely. This library just adds the FreeRTOS scheduling timer (using the WDT), and inserts the Arduino `loop()` function into the FreeRTOS Idle Task. So both Arduino APIs and FreeRTOS APIs are completely as normal.

NoSmoke999, 2021-12-08:
Excellent, and thanks!

closed 2021-12-09 by feilipu