feilipu/Arduino_FreeRTOS_Library #113: define delay() to vTaskDelay()

feilipu, 2022-05-23:
**Describe the issue**
Users expect to use `delay()` which breaks FreeRTOS. So define it to be useful.

**Expected behavior**
Using delay() breaks everything. Redefine it to stop that happening.

**Additional context**
See [discussion here](https://github.com/RalphBacon/244-RTOS-for-Arduino/issues/1).

closed 2022-05-23 by feilipu

technofreak74, 2022-05-29:
Apologies if this is the wrong place to raise this, but I think this change may have broken the platformIO integration.

I'm attempting to build this project: https://github.com/mrwolfjunior/TrigPLC

but I'm getting the following errors building the FreeRTOS code:
```
In file included from .pio/libdeps/controllino_mega/FreeRTOS/src/portable.h:53:0,
                 from .pio/libdeps/controllino_mega/FreeRTOS/src/Arduino_FreeRTOS.h:65,
                 from src/main.cpp:1:
.pio/libdeps/controllino_mega/FreeRTOS/src/portmacro.h:99:47: error: variable or field 'vTaskDelay' declared void
 #define delay(ms)                   vTaskDelay((TickType_t)(ms)/portTICK_PERIOD_MS)
                                               ^
/home/rob/.platformio/packages/framework-arduino-avr/cores/arduino/Arduino.h:143:6: note: in expansion of macro 'delay'
 void delay(unsigned long ms);
      ^~~~~
/home/rob/.platformio/packages/framework-arduino-avr/cores/arduino/Arduino.h:143:12: error: expected primary-expression before 'unsigned'
 void delay(unsigned long ms);
            ^
.pio/libdeps/controllino_mega/FreeRTOS/src/portmacro.h:99:61: note: in definition of macro 'delay'
 #define delay(ms)                   vTaskDelay((TickType_t)(ms)/portTICK_PERIOD_MS)
                                                             ^~
/home/rob/.platformio/packages/framework-arduino-avr/cores/arduino/Arduino.h:143:12: error: expected ')' before 'unsigned'
 void delay(unsigned long ms);
            ^
.pio/libdeps/controllino_mega/FreeRTOS/src/portmacro.h:99:61: note: in definition of macro 'delay'
 #define delay(ms)                   vTaskDelay((TickType_t)(ms)/portTICK_PERIOD_MS)
                                                             ^~
```
Commenting out this #define or rolling back to 10.4.6-3 allows the project to build without error.

feilipu, 2022-05-29:
Try 10.4.6-4 Release.
See #114 discussion for more details.

technofreak74, 2022-05-29:
10.4.6-4 builds without issue! Great timing :smile: thank you!

commit 1cd91fce (2022-05-23): closes #113