feilipu/Arduino_FreeRTOS_Library #114: Arduino delay() changed its behavior to not do busy wait anymore

Prof-Florian-Kuenzner, 2022-05-27:
**Describe the issue**
Since https://github.com/feilipu/Arduino_FreeRTOS_Library/releases/tag/10.4.6-3, the Arduino `delay()` function doesn't do a busy wait anymore. This broke the real-time behaviour on my Arduino Mega because it uses `vTaskDelay()` under the hood, and on the Arduino Mega, there is no real SysTick (here, the watchdog timeout with a resolution of 15 ms is used), what doesn't allow fast task changes during the FreeRTOS scheduler. Furthermore, because of the crazy oscillator, it is also not very accurate (the watchdog oscillator frequency depends (see page 392 on http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega640-1280-1281-2560-2561-Datasheet-DS40002211A.pdf) on the voltage and the temperature). But in combination with ISRs and the direct call of `taskYIELD()` if required, this work well (here, if you are interested what we do: https://www.youtube.com/watch?v=loU-Y_Ek8XI).

**Expected behavior**
The `delay()` function comes from the Arduino framework and not from FreeRTOS. Therefore, it shouldn't be redefined. Or, when necessary, there should be an option where it is possible to decide whether to use the new behaviour or to use the well known "busy wait" behaviour.

**Additional context**
I read already https://github.com/feilipu/Arduino_FreeRTOS_Library/issues/113 and https://github.com/RalphBacon/244-RTOS-for-Arduino/issues/1, but I think redefining well-understood functions from a _non-sleep blocking busy wait_ to a _sleep blocking_ behaviour which breaks the real-time behaviour, where I don't expect changes should be done very carefully.

Maybe we can discuss about that.

feilipu, 2022-05-28:
Hi Florian.
Thanks for taking the time to write a detailed issue.

With respect, I don't think you're the target audience for this library in its default settings. I'm trying to keep the defaults so that a new user can get a sensible outcome, and so that an advanced user can modify what they need to suit their specific application.

When I started seeing people coming in from Ralph's tutorial, and then in discussion he mentioned that he assumed that 'delay()' was redefined to 'vTaskDelay()' by the library as [it is done with the ESP32](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L176) implementation, I went around the internet and found __all__ the YouTube tutorials were using `delay()` incorrectly to blink LEDs.

I could stand on a mountain and shout at everyone _"You're all doing it wrong. You aren't really using FreeRTOS at all, but rather a single priority round-robin scheduler"_. Or, I could just fix it for the vast majority, by defining `delay()` to the sensible default.

---

I'm very interested in the application in your video, but I can't see any code repository to see how you're doing it.

If you're relying on millisecond accuracy (I believe you are), then it would actually be better to use the [Arduino `millis()` function](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring.c#L65) for your timing. This would reduce the overhead imposed by Arduino `delay()` function and would probably get you greater accuracy by omitting its unnecessary decision loops and `long` comparisons which in 8-bit assembly do take time.

Alternatively, you could just comment out my definition of `delay()`, and continue without further change.

I will add some notes in the readme about using `delay()` and `millis()` for timing, to make it clear.

Prof-Florian-Kuenzner, 2022-05-28:
I can feel your pain, seeing all these wrong usages of `delay()`. I mean, this is where it is the difference between fast hacking something, what is really lovely possible within the Arduino framework, and working with real-time systems, where one has to go deeper into the details.

The video with the ball drop example is part of my lecture _Embedded Real-Time Systems_, and the sources are not publicly available right now. The students have to write the code within one day, to get the example working. We practice real-time programming methods and discuss different behaviours, as well as software and hardware limitations of the system.

Yes, in my example, I rely on milliseconds accuracy, which is impossible with the FreeRTOS SysTick on the Arduino Mega, because of the watchdog usage, what gives me a 15ms SysTick that can easily have a jitter of additionally a few ms. Because in this specific code section, only one task is active, and therefore, it is possible to perform a busy wait with enough accuracy. For sure, we could use a hardware timer, but I wanted to demonstrate different possibilities.

For the calculations, we already rely on the `millis()` function, which is driven by `Timer0` behind the scenes. As a workaround, I tested:
- Comment out the `delay()` function
- Undef the macro with `#undef delay` after importing the headers

Both works as a workaround.

Personally, I don't think it's a good idea to change a function from a different library. But if you think this is the best way, then I would add some preprocessor defines where it is easily possible to choose the _default busy wait behaviour from the Arduino Framework_ or the _`vTaskDelay()` driven function from FreeRTOS_. If you like, I can help to implement this.

feilipu, 2022-05-29:
> I mean, this is where it is the difference between fast hacking something, what is really lovely possible within the Arduino framework, and working with real-time systems, where one has to go deeper into the details.

Yes, you've captured exactly the reason why this was written. To enable fast prototyping of a complex system, using existing Arduino Shields and drivers. It started in preparation for a Hackathon.

---

I think my preferred strategy would be to call Arduino `delay(ms)` if the `ms` parameter is shorter than a Tick, and otherwise call `vTaskDelay()` if it is longer. Does that meet your requirement?

I've been trying to think of how to do this with the preprocessor, but I'm coming up with nothing. So, I'll have a go at a `mydelay(ms)` function that will do the comparison and then elect the correct delay type. If I make it an inline function then it shouldn't cause too much extra overhead.

```C
#define delay(ms) mydelay(ms)

void mydelay( uint32_t ms ) __attribute__ ((__always_inline__));

void mydelay( uint32_t ms )
{
    uint8_t milliseconds = (uint8_t) ms;
    if (  (uint8_t) portTICK_PERIOD_MS - milliseconds > 0 )
        delay( ms );
    else
        vTaskDelay( (TickType_t) (ms) / portTICK_PERIOD_MS );
}
```
---

I should also point out that the code using the short `delay()` should be wrapped in a critical section, so that it is not interrupted by the Scheduler (hint for any students reading this 😉). Either using the __avr-libc__ methods in [`<util/atomic.h>`](https://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html) or using the `portENTER_CRITICAL` and `portEXIT_CRITICAL` methods in [`portable.h`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/portmacro.h#L61). Otherwise the ball will be occasionally dropped.

Prof-Florian-Kuenzner, 2022-05-31:
My first idea was to provide something like this:
```c
#define DELAY_BEHAVIOUR_DEFAULT_ARDUINO  (1)
#define DELAY_BEHAVIOUR_DEFAULT_FREERTOS (2)

//here, the user can choice which implementation he wants
//#define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_DEFAULT_ARDUINO
//#define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_DEFAULT_FREERTOS

#ifndef configDELAY_BEHAVIOUR
    #define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_DEFAULT_FREERTOS
#endif

#if configDELAY_BEHAVIOUR == DELAY_BEHAVIOUR_DEFAULT_FREERTOS
    #define delay(ms)                   vTaskDelay((TickType_t)(ms)/portTICK_PERIOD_MS)
#endif
```

Here, it is clear that the intention is to choose, which implementation should be used. What do you think about that?

In your example with the `mydelay()` there is another problem: for small time periods (which depends on `portTICK_PERIOD_MS`) there is a busy wait, while for >= `portTICK_PERIOD_MS` there is the `vTaskDelay()` which can introduce a task change), which is---from a real-time programming view---not very reliable, and should be avoided.

Thanks for the hint with `portENTER_CRITICAL`/`portEXIT_CRITICAL`.

Prof-Florian-Kuenzner, 2022-06-08:
Sorry, I have some busy days. I respond next week when I have time again.

Floessie, 2022-06-09:
@feilipu 

> I could stand on a mountain and shout at everyone "You're all doing it wrong. You aren't really using FreeRTOS at all, but rather a single priority round-robin scheduler". Or, I could just fix it for the vast majority, by defining delay() to the sensible default.

Well, IMHO you should indeed shout when anyone uses `delay()` with the Arduino_FreeRTOS_Library. So why not redefine `delay()` to a compile time error? Or is that too much of a good thing?

Just my 2.3 ¢ (inflation, you know…),
Flössie

leoNavarro95, 2022-06-21:
Hi all!
I think  isn't a good idea to redefine `delay()` to `vTaskDelay()` There are moments where it is necessary to use de default `delay()` behavior for example in an academic environment where I'm currently. I would like to teach to students the difference between using `delay()` and `vTaskDelay()`.

If some profesional dude needs to use correctly FreeRTOS he will use the vTaskDelay aproach anyways.

Greetings from the Caribe

Prof-Florian-Kuenzner, 2022-07-08:
> @feilipu
>
>
> Well, IMHO you should indeed shout when anyone uses `delay()` with the Arduino_FreeRTOS_Library. So why not redefine `delay()` to a compile time error? Or is that too much of a good [...]

You are right, usually, one wants to use `vTaskDelay()`. But here, on the Arduino Mega, the SysTick (that is in this case realised with the Watchdog-timeout) isn't fast enough: This has the drawback, that the real-time requirements can't be satisfied.

Performing a busy wait is -- when you know what you are doing -- is in this case appropriate because with that it is possible to react fast enough and fulfil the real-time requirements.

Prof-Florian-Kuenzner, 2022-07-08:
> Hi all! I think isn't a good idea to redefine `delay()` to `vTaskDelay()` There are moments where it is necessary to use de default `delay()` behavior for example in an academic environment where [...]

I also want to discuss the difference between busy wait (`delay()`) and waiting with a possible task change (`vTaskDelay()`) with my students. I do also recommend a solution where one can choose the behaviour.

Prof-Florian-Kuenzner, 2022-07-08:
@feilipu 
> > Thanks for the hint with `portENTER_CRITICAL` / `portEXIT_CRITICAL`.
> 
> Actually, on further thought this idea to prevent the Scheduler doing an interrupt during `delay()` is not going to be successful.
> 
> The issue is that `delay()` relies on the increase in `micros()` to complete, and with the interrupts globally disabled there can never be a change in [`timer0_overflow_count`](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring.c#L62) and therefore `delay()` will never complete.
> 
> The only solution is to [`wdt_disable()`](https://www.nongnu.org/avr-libc/user-manual/wdt_8h_source.html) disable the WDT from triggering rather than a general interrupt disable, and then restore WDT afterwards.

In my case, the SysTick will not perform a task change because of the task priority design. The task that performs the busy wait with `delay()` has the highest priority and will therefore not be interrupted by any other task. That's why this works in my case.

leoNavarro95, 2022-07-08:
> I also want to discuss the difference between busy wait (`delay()`) and waiting with a possible task change (`vTaskDelay()`) with my students. I do also recommend a solution where one can choose the [...]

Yep, I managed to do that, and also learned about the Tick Interrupt and the Time Slice Algorithm. The thing is that I can't achieve to demostrate to my students about the "busy wait" (i tried with a crude for() loop) and the scheduler preemps the excecution of that "busy wait" this is because of the Time Slice Algorithm with the watch dog timer interrupt that avoid to hapen that in a task. In resume: it is not possible to do a "busy wait" that ocupe all the processing time within a task (while mentioned algorithm been activated wich I recommend to not deactivate)

Prof-Florian-Kuenzner, 2022-07-08:
As an alternative to `delay()`, I just found `_delay_ms()` and `_delay_us()` as stated and the [avr-libc](https://www.nongnu.org/avr-libc/user-manual/group__util__delay.html). @leoNavarro95 this could also be an alternative to teach the busy wait behavior for you.

feilipu, 2022-07-09:
```
//here, the user can choice which implementation he wants
//#define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_DEFAULT_ARDUINO
//#define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_VPORTDELAY
```
Ok, I've done a simple solution, that I think still achieves the required outcomes.

feilipu, 2022-07-09:
@Prof-Florian-Kuenzner 

>> The only solution is to [wdt_disable()](https://www.nongnu.org/avr-libc/user-manual/wdt_8h_source.html) disable the WDT from triggering rather than a general interrupt disable, and then restore WDT afterwards.

> In my case, the SysTick will not perform a task change because of the task priority design. The task that performs the busy wait with delay() has the highest priority and will therefore not be interrupted by any other task. That's why this works in my case.

Actually, I don't think that's going to help in this case. The WDT is still firing, and the interrupt is still being serviced. So if you measure jitter in your timing function, I'm sure you'll see where the WDT interrupted, and was handled by [`vPortYieldFromTick()`](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/port.c#L700), but didn't initiate a task swap. Of course it will be only a few microseconds of jitter, so this may not matter in your application.

Prof-Florian-Kuenzner, 2022-07-09:
> ```
> //here, the user can choice which implementation he wants
> //#define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_DEFAULT_ARDUINO
> //#define configDELAY_BEHAVIOUR DELAY_BEHAVIOUR_VPORTDELAY
> ```
@feilipu 
> 
> Ok, I've done a simple solution, that I think still achieves the required outcomes.

This is exactly what I need. Thanks a lot.

closed 2022-07-09 by Prof-Florian-Kuenzner