feilipu/Arduino_FreeRTOS_Library #116: INCLUDE_vTaskDelayUntil and INCLUDE_xTaskDelayUntil && avr/io.h: No such file or directory

flyingkaktus, 2022-12-22:
**Describe the issue**
I just downloaded your library and am trying to get the AnalogRead example to work. I get the following error message which I don't know how to interpret:

```
In file included from C:\Users\..\AppData\Local\Temp\.arduinoIDE-unsaved20221122-7040-5ioosa.bgo\AnalogRead_DigitalRead\AnalogRead_DigitalRead.ino:1:
c:\Users\..\Documents\Arduino\libraries\FreeRTOS\src/Arduino_FreeRTOS.h:133:10: error: #error INCLUDE_vTaskDelayUntil and INCLUDE_xTaskDelayUntil are both defined. INCLUDE_vTaskDelayUntil is no longer required and should be removed
         #error INCLUDE_vTaskDelayUntil and INCLUDE_xTaskDelayUntil are both defined.  INCLUDE_vTaskDelayUntil is no longer required and should be removed
          ^~~~~
In file included from c:\Users\..\Documents\Arduino\libraries\FreeRTOS\src/Arduino_FreeRTOS.h:1365,
                 from C:\Users\..\AppData\Local\Temp\.arduinoIDE-unsaved20221122-7040-5ioosa.bgo\AnalogRead_DigitalRead\AnalogRead_DigitalRead.ino:1:
c:\Users\..\Documents\Arduino\libraries\FreeRTOS\src/FreeRTOSVariant.h:33:10: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
          ^~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: win11
 - Browser [e.g. chrome, safari]
 - Version Arduino IDE 2.0.3

**Additional context**
Add any other context about the problem here.

gpb01, 2022-12-22:
For which kind of board are you trying to compile?

Guglielmo

flyingkaktus, 2022-12-22:
@gpb01 for an esp32-cam OV2640

feilipu, 2022-12-22:
> an esp32-cam OV2640

Please confirm whether you can see an esp32-cam on the [Compatibility](https://github.com/feilipu/Arduino_FreeRTOS_Library#compatibility) statement.

closed 2022-12-22 by flyingkaktus