feilipu/Arduino_FreeRTOS_Library #118: Arduino Due support

Sylensky, 2023-03-10:
Hello,

i was wondering if it would make sense to include support for the arduino due in this freertos port. Since it is one of the higher sophisticated boards from arduino it would be totally reasonable to have a freertos running on it. Unfortunately there aren't that much port available and if they are quite outdated and doesn't work that well without deep diving into freertos itself. (e.g. https://github.com/bdmihai/DueFreeRTOS)

Previously i used your port on my megas which run perfectly fine but due to hardware limitations iam forced to switch to a higher performant board which in that case is the due until the giga (an absolute monster imo for hobbyists) support is coming around.

I do lack the necessary time to create a port myself which iam not at all familiar with but overall it would be a really nice addition to support that board in this library aswell.
Is it planned in the nearer future or are there quite some limitations for it except of course the time that can be scheduled onto this project.

feilipu, 2023-03-11:
Unfortunately the Due is a bit of an ugly child. Arduino has moved on to other MCUs and the specific ARM MCU variant used in the Due is so old that, well... speaking harshly, no one else is using them.

Putting that aside, the Due is totally unlike the AVR ATmega architecture, so there's no chance that it can be added here.

The best path forward is to use the `port.c` and `portmacro.h`, etc files from the portable directory in the old ports you've found and try them with a modern FreeRTOS release. The port specific files don't change much (if at all) over time or release.

Good luck. Make a repository so other can benefit from your work too. 👍

gpb01, 2023-03-11:
> Am i right to assume that only the `port.c`, `portmacro.h`, `portable.h` are really mcu specific and the rest is freertos specific?

Have a look here: https://www.freertos.org/FreeRTOS-porting-guide.html

Sylensky, 2023-03-11:
> > Am i right to assume that only the `port.c`, `portmacro.h`, `portable.h` are really mcu specific and the rest is freertos specific?
> 
> Have a look here: https://www.freertos.org/FreeRTOS-porting-guide.html

Thanks for the hint. I will take a look.

closed 2023-03-11 by Sylensky