feilipu/Arduino_FreeRTOS_Library #120: lgt8f328p compatibility

mchacher, 2023-08-03:
Thanks for this great work. 
I would like to use FreeRTOS library with lgt8f328p from LogicGreen. Is this micro supported by chance?
thanks a lot.

feilipu, 2023-08-04:
If the lgt8f328p is 100% compatible with the ATmega328p then there should be no problems. Just install the board support packages and load the library as usual.

But I'm thinking that there may be some issues with the WDT triggering, if it is not actually 100% compatible, as noted [here](https://www.eevblog.com/forum/projects/anyone-here-interested-in-the-logic-green-avrs-lgt8f328p/msg2785838/#msg2785838). It may just be a fuse setting issue for the individual boards, which can be corrected in `boards.txt`, but it may be a hardware implementation issue. And if so then one of the other Timers will need to be configured to to provide the System Tick.

feilipu, 2023-08-17:
Any luck? If so, please let me know.

gpb01, 2023-08-17:
I have noticed that the lgt8f328p has four timers (_one more than the ATmega328P_), so it might be possible to use **Timer3** (_8 bit timer_) for tick generation ... 

Guglielmo

feilipu, 2023-08-18:
Yes. Did not know that.

Further examples at [avrfreertos](https://github.com/feilipu/avrfreertos), show how Timer0 (an 8 bit Timer) can be configured for ATmega328p. Timer3 will probably be no more difficult than exchanging 0 for 3 in the code in this case.

Note that ATmega1284p Timer3 is 16 bit so is not a good example here.

mchacher, 2023-08-18:
thanks all for your feedback. Not sure I am following. Finally what would be your recommendation to make it work? I confirm that the code as it is not working.

feilipu, 2023-08-18:
[Here](https://github.com/feilipu/avrfreertos/blob/master/freeRTOS10xx/portable/port.c#L57) are the constants for Timer0 which may be the same as Timer3.

Here is the interrupt configuration. And here the interrupt needs to be adjusted too.

Basically read the FreeRTOS10 code and replace Timer0 with Timer3.

gpb01, 2023-08-20:
Regarding my previous post, upon closer inspection of the various "_translations from Chinese_" of the datasheet, it appears that Timer 3 of the LGT8F328 **is a 16-bit timer**, not an 8-bit timer.

From the datasheet:

>Overview
TC3 is a general purpose 16-bit Timer/Counter module, with PWM support. It allows accurate wave generation.

Guglielmo

feilipu, 2023-08-23:
Good to hear. I suggest following the conventions in avrfreertos `port.c` for naming the System Tick source.

For example `portUSE_WDTO` is used for the standard WDT. Perhaps use `portUSE_WDT_LG328` or similar.

These definitions are selected in `FreeRTOS_Config.h`.

feilipu, 2023-08-24:
I'm proving `avrfreertos` as an example or reference, as the kind of thing you are trying to achieve has been done there already. Specifically alternate MCUs, alternate boards, and alternate Timers. Just treat it as reference material.

`Arduino_FreeRTOS` is simple by design with one Timer setup and all the potential alternatives and definitions are not present. So there's no examples there to see, unfortunately.

In an Arduino Library, it also difficult to make things configurable. And all the MCU resources are normally utilised by Arduino, so it becomes quite tricky not to break something a new or less technical user would expect to "just work".

feilipu, 2023-08-24:
I've pushed [an update](https://github.com/feilipu/Arduino_FreeRTOS_Library/releases/tag/10.5.1-1) that might make things work, provided the [board definitions](https://github.com/dbuezas/lgt8fx) are installed correctly.

I'm a bit confused at to what the most generic term for the LGT8F328P is. It is referred to in the variant files under different definitions, and there doesn't seem to be a hierarchy present so I went with below.

```c
#if defined(__LGT8FX8P__) || defined(__LGT8FX8E__) || defined(__LGT8FX8P48__)
#endif
```

gpb01, 2023-08-28:
Hi Phillip,
just a point of clarification ... looking at the detail of the compilation string used by core of "dbuezas" I see:

`-mmcu=atmega328p -DSERIAL_RX_BUFFER_SIZE=64 -DCLOCK_SOURCE=2 -DF_CPU=(16000000L/1) -DF_OSC=16000000L -DF_DIV=1 -DARDUINO=10819 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_AVR`

so ... no reference to the id referenced in your previous post (_maybe they are automatically defined by the compiler_). 

In any case, since the change to WDT **is only present in the core of "dbuezas"** and not in other cores for LGT8F328P, I suggest using:

```
#ifdef ARDUINO_AVR_LARDU_328E
...
#endif
```

...what I believe is unique to the above core :-)

Guglielmo

feilipu, 2023-08-28:
Thanks, I'll modify to use that definition soon.
I also need to add the `lgtwdt.begin()` function to the `startup()` code too.

It would be ideal if your core WDT improvement PR could define similarly named functions for use of the WDT as are in use with the ['avr-libc/wdt.h`](https://www.nongnu.org/avr-libc/user-manual/group__avr__watchdog.html) file and the additional [interrupt function](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/port.c#L83) and [interrupt+reset function](https://github.com/feilipu/Arduino_FreeRTOS_Library/blob/master/src/port.c#L149) I've added, that way only the correct header would need to be selected and defined, rather than needing to define alternate functions too.

I don't know if that would create namespace issues, but I think not.

gpb01, 2023-08-30:
Hi Phillip,
I looked the example you provide for Timer 0 and ...

1. I did not understand how and who includes the .cpp file you provide (tick_sources_timer0.cpp) in the project.

2. In the above file I see the function "prvSetupTimerInterrupt", I see the definition of the ISR, but I do NOT see:

2.1 vPortEndScheduler ... which, obviously, in the case of using a Timer, must be provided.

2.2 I don't understand how to connect the ISR defined in "tick_sources_timer0.cpp" with the various conditions where "configUSE_PREEMPTION" is 0 or is 1 (_different things need to be done inside the ISR_)

Can you provide clarification ...

Thank you,

Guglielmo

P.S.: Another strange thing ... If I compile the program that I indicated [HERE](https://github.com/feilipu/Arduino_FreeRTOS_Library/issues/120#issuecomment-1695888170) everything is fine and there are no errors; if I try to include Timer 3 related definitions directly inside "FreeRTOSVariant.h" or inside "port.c" I get errors as if it no longer recognizes the existence of Timer 3 ... what am I doing wrong?

feilipu, 2023-08-30:
> Now I have an Arduino_FreeRTOS library working on LGT8F328 using Timer 3 for ticks :-)

Good news.

Please prepare a PR so I can see exactly how. You did it. I will use that as a guide, and further clarify the instructions for next time.

I'm AFK until late next week, so no rush. So I'll sort it when I'm back.

gpb01, 2023-09-12:
PR is waiting ... :-)

Guglielmo

feilipu, 2024-01-02:
Closed by https://github.com/feilipu/Arduino_FreeRTOS_Library/pull/121

closed 2024-01-02 by feilipu