feilipu/Arduino_FreeRTOS_Library #122: Arduino R4

KenwoodFox, 2023-10-20:
I know this is maybe a longshot but, im trying to port my code to the R4 because of memory issues on my regular uno R3.

FreeRTOS worked super well on the R3, but i just need more features, more tasks, and i just simply didn't have enough RAM on the old platform.

Im trying to build Arduino freertos but obviously, there's no avr.h on the renesas chip, and my compiler fails right there.

```
Compiling .pio/build/uno/lib195/FreeRTOS/port.c.o
In file included from .pio/libdeps/uno/FreeRTOS/src/Arduino_FreeRTOS.h:64,
                 from .pio/libdeps/uno/FreeRTOS/src/event_groups.c:38:
.pio/libdeps/uno/FreeRTOS/src/FreeRTOSConfig.h:32:10: fatal error: avr/io.h: No such file or directory
   32 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
```

Is there any way I can help maybe, include support for the other arduino chips? I'd like to be able to keep using freertos in my project. Any advice or how i could get started would be super helpful thanks.

feilipu, 2023-10-20:
It is a very relevant question. From the start, I've been clear that this repo is to enable people to learn the FreeRTOS APIs and code in a very simple environment. Most of the complexity and additional code has been stripped away, leaving only the smallest number of files possible to get started.

That is useful when the basic Arduino [Decimila](https://docs.arduino.cc/retired/boards/arduino-diecimila) and later 328p UNO was the natural first target, but now that that is changing it brings the purpose of this repo into question.

And, as I get the most joy out of writing [assembly language with 8 bit processors](https://github.com/z88dk/z88dk/graphs/contributors) currently, it is unlikely that I'll be writing a 32 bit port for the Arduino UNO R4 any time soon.

However, there are still two options that you can pursue.

Firstly, you can use one of the larger AVR devices, if the issue is only memory. The MEGA is supported, as are all of the intermediate devices. I ran into memory issues some years ago with some larger projects and built my [Goldilocks](https://www.tindie.com/products/feilipu/goldilocks-analogue/) based on a ATmega1284p to make them successful. All the design files are public, so you can get one made to suit your exact requirements if you prefer.

For the Arduino UNO R4 there is already a FreeRTOS port hidden within the Renesas RA4M1 [Flexible Support Package](https://www.renesas.com/us/en/software-tool/flexible-software-package-fsp#additional_details). If you read the source files you should get a pretty good idea of how to strip them out to make an Arduino library similar to this one. As you will have also seen it is just a [Cortex M4 which is already supported by FreeRTOS](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/GCC/ARM_CM4F). So it is not a complex job to build what you need.

KenwoodFox, 2023-10-21:
Thats a really good response! i appreciate you taking the time to give me all that helpful information!

I knew it would not be easy to do the port here, I figured i would try and offer. But its very handy to know that there are some other options out there! I guess i should check out the FSP as platformio supports it?

I was hoping I could maybe try and port it myself somehow, i know there already is a freertos for that type of processor the cortex m4 but, i didn't know how to include it as a library or if it would work with the arduino enviorment like yours. Everyone says i should move away from Arduino but, im just trying to port and not rewrite everything. And the mega is a good suggestion! but its physically larger so, might not work out for me at the minute.

Thank you again! I'll try and look into FSP and maybe i can make my own adapter for the cortex m4?

closed 2023-10-21 by KenwoodFox

gpb01, 2023-10-21:
>_For the Arduino UNO R4 there is already a FreeRTOS port hidden within the Renesas RA4M1 ..._

If you look inside the Arduino core libraries for the UNO R4 series, you will see that the FreeRTOS library is already included.

/packages/arduino/hardware/renesas_uno/1.0.4/libraries/Arduino_FreeRTOS

Guglielmo

feilipu, 2023-10-21:
Perfect!

KenwoodFox, 2023-10-21:
Omg, @gpb01 is totally right. I didn't even think it would be so easy XD

I added a single .patch to enable the features feilipu had here and.. it just built! Thank you!