feilipu/Arduino_FreeRTOS_Library #145: Support static allocation

dvescovi1, 2025-05-16:
I see the configSUPPORT_STATIC_ALLOCATION (which was new in later versions of FreeRTOS) is not defined by default in Arduino_FreeRTOS.h.
The comment is "Default to 0 for backward compatibility". I may be wrong but I do not see any backward compatibility issue with enabling this as all static allocation functions are unique. Except for the slight library size increase. 
The dynamic allocation can still be left enabled as static and dynamic allocation are not mutually exclusive.
Allowing static allocation is very important especially with resource constrained devices like the Arduino's. Its important to keep close tabs on memory allocation and execution speed.

feilipu, 2025-05-17:
There are some issues with supporting static allocation by default. Some additional functions need to be provided.
See in the [documentation here](https://freertos.org/Documentation/02-Kernel/03-Supported-devices/02-Customization#configsupport_static_allocation), and there is a broader discussion of the [pros and cons of memory allocation here](https://freertos.org/Documentation/02-Kernel/02-Kernel-features/09-Memory-management/03-Static-vs-Dynamic-memory-allocation).

These call-back functions for static memory allocation were added in #19, and are available to use, by just setting the `configSUPPORT_STATIC_ALLOCATION` to `1`

So it remains open for the advanced user to configure this library in any way they like.
However static memory allocation is not something that should concern a new user, so it has been left off by default.
Please just turn it on after installing it on your Arduino IDE, if you prefer.

feilipu, 2025-05-17:
No, you'll need to change it in the configuration file in the actual installed library. It is an unfortunate limitation of the Arduino IDE. You can find info on where the files are installed for your platform by any search engine.

But the library is recompiled (almost) every time you start the IDE so that's not really a problem in practice.

The only inconvenience is that you have to edit inside the installed library files with another editor, and you can't really do that with the IDE (at least to my knowledge).

With that knowledge you can tune the configuration in other ways too. For example substitute the memory allocation solution for a fixed allocation, which will support stack and heap tracking. `heap_1.c` or `heap_2.c` can be used if you have a specific target in mind. See my other AVR repositories for details.

dvescovi1, 2025-05-18:
thanks for the feedback

closed 2025-05-18 by dvescovi1